StefanSchippers/xschem #225: tcleval() question

georgtree, 2024-07-12:
Hello!
I tried to use tcleval(), and for text on schematic it works only if I add name into the props like this:
[image]
If I remove it, I get:
[image]

The issue here is that I want to use tcl substitution for text of the symbol, that is shown only if particular variables is set.
Thank you in advance, George.

StefanSchippers, 2024-07-13:
to enable the tcl substitution either use `name=inst_name` so you can also use the `@var` variables related to the specified instance (like `@symname`, `@name`, `@symref`, `@#PLUS:spice_get_voltage`) or If you just need tcl evaluation put attribute `floater=1` in the props textbox.

The reason is to avoid doing tcl substitutions on all texts since this step is slower.

georgtree, 2024-07-13:
I got it, thank you!

closed 2024-07-13 by georgtree