StefanSchippers/xschem #228: a netlist problem, and selecting from right to left dont select the touching things anymore

olisnr, 2024-08-13:
dear stefan

i have a problem generating a spice netlist from `tst/aa_mirrors_biastest_04.sch` in the newest version. it produces an error 
```
-----------/ALL/Xschem/design/mirrors/aa_mirrors_biastest_04.sch
Error: undriven node: #net71
Error: undriven node: #net72
Error: undriven node: #net73
-----------/ALL/Xschem/design/opamp_hv_bias_01.sch
-----------/ALL/Xschem/design/opamp_hv_bias_03.sch
-----------/ALL/Xschem/design/mirrors/aa_mirrors_biastest_04.sch
```
but if i search in the text-editor for net71, there is no one. ican You look at it? how can i debug this?

and i would like how to enable the "right to left selecting touching things" mode again. i didnt found ir in the `xschemrc`.

[tst.tar.gz](https://github.com/user-attachments/files/16598708/tst.tar.gz)

thanks a lot

StefanSchippers, 2024-08-14:
> and i would like how to enable the "right to left selecting touching things" mode again. i didnt found in the xschemrc.
Please add/change  `set incremental_select 1` in yor xschemrc.

StefanSchippers, 2024-08-14:
In case of all-disconnected components (there is no  pin connected to something) print an error message in the ERC window and highlight them. Pressing the '5' toggle after the netlist operation allows to easily find them.
if you have a chance please test with latest xschem build.
[image]

olisnr, 2024-08-14:
super! thanks! 

do You have also an idea why the .raw files arent loaded/updated in the newest version of Xschem? it seems that one of the many changes, since i last time downloaded it has broken or improved something :)

if i use the script-button to simulate and load the .raw files it only simulates, but dont load, it did in the old version automatic after the simulation script. now it looks like this:

run
[image]

 load
 [image]

simulate dont update plots
[image]

open a plot dialog shows data
[image]

closing the dialogbox updates a single plot - there are others that also have something to show but arent updated.
[image]

olisnr, 2024-08-14:
the simulation script is:

```
name=h13 
descr="netlist simulate corners ALL" 

tclcommand="

    proc simulate_and_plot {corners load_the_plots} {

	global netlist_dir
	global execute
	global mos_corner

        set schematic [file rootname [file tail [xschem get schname 0]]]

        foreach corner $corners { # write different netlists
            set mos_corner ${corner}; # export for NGSPiCE TCL
            xschem netlist ${netlist_dir}/${schematic}_mos${corner}.spice
        }

        puts netlists_written;

        foreach corner $corners { # simulate all netlists
            xschem set netlist_name ${schematic}_mos${corner}.spice
            set id_${corner} [simulate ]
        }

        puts sims_started;

        while {true} {
            set info_exists false;
            foreach corner $corners {
                set mos_corner $corner
                set test_cmd [string cat {execute(cmd,} [set [string cat {id_} ${corner}]] \)]
                if {[info exists $test_cmd]} {
                    set info_exists true;
                }
            }
            if {!$info_exists} {
                puts sim_ended;

                break;
            }
            puts -nonewline .; flush stdout;

            delay 99;
        }

        if {${load_the_plots}} {
            xschem raw_clear;
            xschem annotate_op ${netlist_dir}/${schematic}_mostt.raw
        }

        puts done

    }

    simulate_and_plot ${corners} true
"

```

olisnr, 2024-08-14:
i tried the _Please add/change set incremental_select 1 in yor xschemrc._ and it does the "left-right selection", but also the "add but not remove". if its possible i only want the "left-right selection" with `incremental_select 0`, like it was working in the old version.

StefanSchippers, 2024-08-14:
The `xschem raw file type` command now does not redraw the graph with the new data. A `xschem redraw` command is needed for that.

The alternate command `xschem raw_read file type` on the other hand does a load + redraw of the waveforms.

olisnr, 2024-08-14:
> The `xschem raw read rawfile sim_type` command now does not redraw the graph with the new data. A `xschem redraw` command is needed for that.
> 
> The alternate command `xschem raw_read rawfile sim_type` on the other hand does a load + redraw of the waveforms.

this commands dont work either. the problem seem to be, that Xschem dont look in the diagrams anymore what to load. it loads the data when i access the plots dialog.  

it could be a problem, that i dont generate a data-file like `[file rootname [file tail [xschem get schname 0]]].raw` but different versions of `[file rootname [file tail [xschem get schname 0]]]_mos${mos_corner}.raw`. so there is no "standard-raw file" avaible...

olisnr, 2024-08-14:
it has nothing to do with the non-standard filenames. also the `Waves-button` right in the menue-bar dont work, if i load a raw-file manually.

StefanSchippers, 2024-08-14:
Please also note that recent xschem has an `Auto Load` checkbutton in the graph dialog box. 
If this checkbox is disabled graphs will not automatically load the raw files specified therein. Please enable it. Some other users were complaining about the automatic loading.
[image]

olisnr, 2024-08-14:
i downloaded the latest source and my dialog looks like this:
[image]

do You use an old version :)

olisnr, 2024-08-14:
it seems that i need also to `"sudo make install"` to update the dialog. now i see the same dialog like You

StefanSchippers, 2024-08-14:
if the installation goes to a system directory (like /usr/local, the default unless changed in configure by the --prefix option) you need to install with root privileges, so sudo is needed. Check to see if there are multiple xschem installations in different places, this is almost certainly a source of problems.

Latest xschem build now has a xschemrc variable `select_touch`(default=1)  that enables selecting objects "touched" by the selection rectangle if dragging to the left. This behavior is now separated from `incremental_select` variable (default=1) that enables showing selected  objects while dragging.

olisnr, 2024-08-14:
cool: everything is working again! thanks a lot

closed 2024-08-14 by olisnr