StefanSchippers/xschem #229: a selection question

olisnr, 2024-08-15:
if i select a part with `shift` pressed, i can add the part to the selection. but would like to toggle the selection while pressing `shift`. as example to select all pmos per `cnrt-f` and then deselect some of them via `shift click`.
is there a xschemrc parameter to enable this?

olisnr, 2024-08-19:
and a 2nd selection question: if i want select all symbols `opamp_diff_bi4.sym` what should i put in the "Token" field of the search dialog?

StefanSchippers, 2024-08-19:
Unselection can be done by: 

1. `Alt-Click` an object (left button press with `Alt` key pressed)
2. Drag an **unselection** rectangle with the mouse by keeping the `Alt` key pressed.

Selecting all the schematic objects (nets, texts, symbols , etc) can be done by pressing `Ctrl-A`

Selecting all symbols can be done in various ways. One accurate way is:
setting token to `cell::name`
setting value to `\.sym$` (this regular expression means 'names ending with `.sym`)
(however see below for a simpler search option).

In the token field the prefix `cell::xxxx` finds attribute `xxxx` in the symbol, while writing just `xxxx` as token will look for attribute `xxxx` in the instance.

For example `name` is the instance name of a symbol (`D1`, `M1`, `X1`, ...) while `cell::name` is the name of the symbol, like `vsource.sym`, `res.sym` and so on.

If you leave the `value` field empty xschem will find all objects that **have** the specified attribute, regardless of the value. So in the above symbol search you could also leave the value field empty, while searching for `cell::name`, so all symbols that have a `name=...` attribute will be highlighted.

StefanSchippers, 2024-08-19:
To select all opamp_diff_bi4.sym symbols search for:
```
token: cell::name
value: opamp_diff_bi4.sym
```

StefanSchippers, 2024-08-20:
Are you using a native linux machine or through some emulation layer, like wsl/wsl2/VirtualBox/VMware?

Emulators often mess up with keymaps and change things. Some Virtual machines intercept the Alt key for some VM specific actions.

StefanSchippers, 2024-08-20:
you can use the `xev` utility, point the mouse inside the `xev` window, press and release the `alt` key and send me the result. I get this:
```
KeyPress event, serial 33, synthetic NO, window 0x2400001,
    root 0x43b, subw 0x0, time 1217849, (99,110), root:(176,275),
    state 0x0, keycode 64 (keysym 0xffe9, Alt_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 36, synthetic NO, window 0x2400001,
    root 0x43b, subw 0x0, time 1217969, (99,110), root:(176,275),
    state 0x8, keycode 64 (keysym 0xffe9, Alt_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

olisnr, 2024-08-20:
i have `Linux Mint 21.2` (based on `ubuntu 22.04`) direct on a notebook.
my `alt-key` gives me:
```
KeyPress event, serial 28, synthetic NO, window 0x6800001,
    root 0x787, subw 0x0, time 438190011, (103,90), root:(2355,444),
    state 0x0, keycode 64 (keysym 0xffe9, Alt_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False
```
my keyboard layout is `swiss german`.

StefanSchippers, 2024-08-20:
Can you please also post the key release? this is where the `state` field is set (on my system it is 0x8)

StefanSchippers, 2024-08-20:
Also check if `Alt+Left mouse button click` is used by the window manager to do some operations (like move a window without clicking the top bar or other actions). In this case the Alt-click event is not passed to the underlying application but "eaten" by the window manager.

olisnr, 2024-08-20:
```
KeyPress event, serial 28, synthetic NO, window 0x6200001,
    root 0x787, subw 0x0, time 440317695, (75,94), root:(2045,176),
    state 0x0, keycode 64 (keysym 0xffe9, Alt_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 28, synthetic NO, window 0x6200001,
    root 0x787, subw 0x0, time 440317935, (75,94), root:(2045,176),
    state 0x8, keycode 64 (keysym 0xffe9, Alt_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

StefanSchippers, 2024-08-20:
Thank you. All the key codes for the left-Alt key atre identical. 
try to see if Left-Alt + Left mouse button click is used by the window manager (try to open its configuration windows and see if Alt + mouse is used to do some thing.

Also test if you can use the Windows Key instead of Alt.
Xschem also recognizes The Win key and used the same way as left-Alt (this is necessary for example on MACs that do not have an Alt key AFAIK).

olisnr, 2024-08-20:
there are no `alt+mouse` bindings in the cinamon hot-keys. but on this machine works the windows key. thanks! 
i will test it tomorrow also on the other device. i was thinking i tested all keys...

olisnr, 2024-08-20:
`cntr-alt` is also working:

```
KeyPress event, serial 25, synthetic NO, window 0x6200001,
    root 0x787, subw 0x6200002, time 442104595, (43,66), root:(1085,314),
    state 0x0, keycode 37 (keysym 0xffe3, Control_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 25, synthetic NO, window 0x6200001,
    root 0x787, subw 0x6200002, time 442104595, (43,66), root:(1085,314),
    state 0x4, keycode 64 (keysym 0xffe9, Alt_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 28, synthetic NO, window 0x6200001,
    root 0x787, subw 0x6200002, time 442105197, (43,66), root:(1085,314),
    state 0xc, keycode 37 (keysym 0xffe3, Control_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 28, synthetic NO, window 0x6200001,
    root 0x787, subw 0x6200002, time 442105198, (43,66), root:(1085,314),
    state 0x8, keycode 64 (keysym 0xffe9, Alt_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

the `windows-key` isnt that great, because he opens also the start-menu.

StefanSchippers, 2024-08-20:
> cntr-alt is also working:

I still suspect that Alt+Mouse button is intercepted by some higher level tool and not presented to the application.

Indeed, see this thread:
 https://forums.linuxmint.com/viewtopic.php?t=247748
where another user needs to "free" the Alt+Mouse button combinations.

olisnr, 2024-08-21:
i searched in `gconf-editor` but found non `Alt+Mouse`. there is a `Alt+Space`, but it has no influence.
would it be complex to make `shift-l_clkick` toggeling the selection?

StefanSchippers, 2024-08-21:
Shift + click is already used to add objects to the selection.

StefanSchippers, 2024-08-21:
so what happens if you do Alt + Left button click on an object? does it get selected as with a normal left button click?

StefanSchippers, 2024-08-21:
If you do Alt + Left button click on another window and then move the mouse, does the window move?

StefanSchippers, 2024-08-21:
The internet is literally full of _furious_ users complaining about cinnamon "catching" Alt+Mouse button events, preventing these sequences to be processed by applications.
See this other thread:
https://forums.linuxmint.com/viewtopic.php?t=264172

olisnr, 2024-08-22:
thanks a lot: that was the problem: the `Alt key` was used to move the window. 

[image]

from a users-perspective i would prefere a toggeling selection. as example QUCS uses `cntr+Lmouse` for toggeling the selection while inkscape uses `shift+Lmouse` . this is really easy to use, You can add and remove parts without searching an othere modifier key on the keyboard.

olisnr, 2024-10-14:
after some weeks doing other stuff, i forget the "change order shortcut `shift-S`". 

could we have the command also in the `Properties Menue`?

StefanSchippers, 2024-10-14:
Done. Thank you for reporting this missing menu item.
Properties -> Change selected object insertion order

closed 2024-10-22 by olisnr