StefanSchippers/xschem #232: How can I fix that?

trongcuhp, 2024-09-11:
[image]

StefanSchippers, 2024-09-11:
If you plan to use Xschem on windows go to this URL and read carefully the README_WIN and INSTALL_WIN files.

trongcuhp, 2024-09-12:
thank for your help

trongcuhp, 2024-09-12:
[image]
I have installed fully but still get this error, do you have any way to run on win like INSTALL_WIN

trongcuhp, 2024-09-12:
> If you plan to use Xschem on windows go to [this URL](http://cgi.repo.hu/cgi-bin/minisvn.cgi?cmd=browse&repo=xschem&path=trunk/XSchemWin) and read carefully the README_WIN and INSTALL_WIN files.

help me!

StefanSchippers, 2024-09-12:
I don't have windows, so my help is limited. 
Read carefully the README_WIN file and build the program with VisualStudio 2022.

StefanSchippers, 2024-09-12:
You can get a pre built windows package by downloading [XSchem-3.4.5-64bit.msi](http://repo.hu/projects/xschem/releases-bin/XSchem-3.4.5-64bit.msi) and installing it on your machine.

trongcuhp, 2024-09-12:
> You can get a pre built windows package by downloading [XSchem-3.4.5-64bit.msi](http://repo.hu/projects/xschem/releases-bin/XSchem-3.4.5-64bit.msi) and installing it on your machine.

can u give me links, I cant download file that u just gave me

trongcuhp, 2024-09-13:
or can you make videos instruct to build on window with micro visual studio,

StefanSchippers, 2024-11-28:
If you can not download this installer file it is probably a windows protection issue.

closed 2024-12-25 by StefanSchippers