StefanSchippers/xschem #233: an ammeter.sym that shorts in LVS export

olisnr, 2024-09-11:
it would be interesting if the ammeter in the LVS export shorts 2 nets. i tried to do this, but it dont work correct, because the netlist-generator detects the short, and dont want to produce the subckt.

is there a way around it? and that is the device short.sym for?

[ammeter_lvs.zip](https://github.com/user-attachments/files/16965910/ammeter_lvs.zip)

StefanSchippers, 2024-09-11:
Here another clip showing how to use the short.sym device.

https://github.com/user-attachments/assets/b03d3f5f-4f66-42c5-9b1d-c07df660a62d

olisnr, 2024-09-11:
thanks a lot!

is there also a solution to replace spice parameters like` l={15*lp}` with the calculated values in CDL netlists? 
i am thinking about using ngspice in debug-mode to generate a intermediate-netlist and use it for LVS. what do You think?

olisnr, 2024-09-11:
is it also possible to short different label or do i need to change the schematics? 

https://github.com/user-attachments/assets/4d06e844-efb6-42df-867e-7083261d2072

StefanSchippers, 2024-09-11:
You can use conditional labels. A label or GND/VDD tier becomes say VSSI or VSS depending if we are doing a LVS netlist or a simulation netlist.

Uploading conditional_lab.mp4…

StefanSchippers, 2024-09-11:
> is there also a solution to replace spice parameters like l={15*lp} with the calculated values in CDL netlists?

If lp is a spice paramer (.param lp = ...) then the calculation needs to be done by ngspice. Xschem can do nothing about that. You can use a tcl parameter for this, if this makes sense for your case.

https://github.com/user-attachments/assets/f2df73f9-171e-4be2-80fb-2cb489ab9c9f

olisnr, 2024-09-13:
i would like to use the TCL solution. and because SPiCE dont need the `$` in the `{}`-expression, i try `w="tcleval([string map {{$} {}} [ev {11*$pw}]])"` 

now comes the next difficulty :) the propagation over the parent-schematic. when i set parameters in the OPAMP-symbol, like:
```
name=x5 
nw="tcleval([string map {{$} {}} $nw])" 
nl="tcleval([string map {{$} {}} $nl])" 
pw="tcleval([string map {{$} {}} $pw])" 
pl="tcleval([string map {{$} {}} $pl])" 
schematic="tcleval($opamp)"
```
the OPAMP shows the correct parameters (2u/1u), but the sub-sheet dont get it. could it be, that there is a trigger missing in the `e`-command? or does it work somehow different?

olisnr, 2024-09-13:
i think, one problem is the 1u i need to try 1e-6