StefanSchippers/xschem #234: symbols & schematics in a project directory

olisnr, 2024-09-13:
i would like to make a directory, for a few hierarchical `.sch` and there `.sym`. but i dont find the way, how to find the symbols, without defining an absolute path, or add the directory in to the `xschemrc`.

i tried also to use `tcleval {}` in the `Edit Properties` dialog box of the `---MISSING SYMBOL---`

StefanSchippers, 2024-09-13:
Create a directory "my_directory": /path/to/my_directory
Put symbols and their corresponding schematics in that directory
in xschemrc: append XSCHEM_LIBRARY_PATH :/path/to/my_directory

the XSCHEM_LIBRARY_PATH is the only variable that holds a list of absolute paths that are system specific. All the schematics should only reference relative paths (relative to the XSCHEM_LIBRARY_PATH paths)

olisnr, 2024-09-13:
thanks
the thing is: i would like not to manipulate the `xschemrc`. but be able to specify a `symbol_file.sym` relative to the schematic, that wants to draw it. 

the `TOP.sch` should be the in the "pwd". i could also use some TCL code to get the absolute path of the symbol-file, by manipulating the schematic path like: 
`[file dirname [xschem get current_name]]/OTA33_BiAS.sym`, but i cant put it in the dialog box of the symbol, because it would produce a path like: `/home/ich/"tcleval{${sr...`.

is it possible to set the symbol-path as text using `tcleval {}` in the dialog? like the path i can specify for an alternative schematic. as example: `schematic="tcleval($opamp)"`
is this also possible for the symbol?

StefanSchippers, 2024-09-13:
You can put tcl expressions as tcleval(_<<<some_script_that_returns_symbol_path>>>>_)

StefanSchippers, 2024-09-13:
see this short clip

https://github.com/user-attachments/assets/8e66671e-5a64-4168-aaaf-f74c58c92d67

olisnr, 2024-09-14:
thanks a lot. now the `e-command` is working, but for the netlist-generation, there also a need for a `schematic="tcleval([file dirname [xschem get current_name]]/OTA33.sch)"`. otherwise its missing in the .spice-file.

is this intendned?

https://github.com/user-attachments/assets/ad80702b-b664-4fe3-9ce0-629fe9651f0d

StefanSchippers, 2024-09-14:
If the subschematic associated to the symbol has the same root name of the symbol and is in the same directory of the symbol there is no need to use a schematic attribute at all. If this does not work for you please send a small test case showing the failure. Anyway it is always possible to use tcleval(...) in a schematic attribute, it will be evaluated and the resulting schematic will be used.

StefanSchippers, 2024-09-14:
I have made a commit, as you noticed the load symbol definition function in xschem (load_sym_def()) did not process a tcleval() if present, so that symbol was not present in netlist.

However the method of using the dirname of the containing schematic to determine the path of the symbol is flawed and will not always work as intended when netlisting a circuit.

The same symbol could be used in different schematics at different hierarchy levels, perhaps located in different places in the filesystem, so each instance of the symbol will be searched in a different directory.

This is not how netlisting works.

The netlister **does not descend into each branch of the hierarchy**. 
Netlisting starts by writing the top level netlist. All used components are saved in a "component" list.
Then loop through each used component that is a `type=subcircuit`, load the corresponding subcircuit schematic and netlist the subcircuit. Add any used components to the "component" list if not already there. At the end all used subcircuits are netlisted one and only one time (a subcircuit might be instantiated in multiple places, but is netlisted only once).
the `[file dirname [xschem get schname]]` _while netlisting_ returns the path of the top level schematic, not the path(s) of the schematic(s) containing the symbol. So this trick works only if the top level is in the same directory of the subschematic(s) containing the symbol referenced with `tcleval([file dirname[xschem get schname]]/sym_name.sym)`.
One last comment: to get the absolute path of current schematic use `xschem get schname`, not `xschem get current_name` (which returns just the relative filename if the base directory is in `XSCHEM_LIBRARY_PATH`).

**xschem get schname : get full path of current sch.
xschem get current_name: name of current design (no library path)**

StefanSchippers, 2024-09-14:
even better: `xschem get current_dirname` will return the absolute path of the directory containing the current schematic.

olisnr, 2024-09-16:
thanks a lot.
somehow its still not working in my schematic. im to stupid to make it work:

[image]

with schematic="tcleval([xschem get current_dirname]/OTA33.sym)" << .sym instaed of .sch is also working :)

[image]

the test-case is OTA33_test.sch 
[OTA33.zip](https://github.com/user-attachments/files/17010979/OTA33.zip)

olisnr, 2024-09-16:
[OTA33.zip](https://github.com/user-attachments/files/17011204/OTA33.zip)

olisnr, 2024-09-16:
thanks a lot

unfortunately netlisting is still not working with from the schematic `OTA33_test.sch` 
but it works from the schematic `OTA33.sch`.
i'm a little at a loss. i dont see the difference.

and there is a new error if the `Edit Properties` dialaog of the OTA33 in `OTA33_test.sch` inclueds `schematic="tcleval(tcleval([xschem get current_dirname]/OTA33.sch))"`

[image]

[OTA33.zip](https://github.com/user-attachments/files/17011505/OTA33.zip)

StefanSchippers, 2024-09-16:
> unfortunately netlisting is still not working with from the schematic OTA33_test.sch
> but it works from the schematic OTA33.sch.

You need to go into OTA33.sym and remove or comment the `default_schematic=ignore` attribute.

olisnr, 2024-09-16:
thanks!!!
i was searching everywhere but not in the .sym... i hope i remember next time!

closed 2024-09-16 by olisnr