StefanSchippers/xschem #235: question about LVS export

olisnr, 2024-09-17:
i do an .cdl export for LVS, for a MOSFET with w=16µm and ng=9, and that gives an effective w of 15.975µm in the layout. this produces an LVS error. if i change the value in Xschem the netlist-export rounds it to 15.97µm, and this gives me still an error :)

the grid of the iHP-openPDK is 5nm. could You export that precision?

StefanSchippers, 2024-09-17:
Please check the netlist. If I set w=1.2345678 on a transistor in xschem the netlist prints w=1.2345678. Probably some other tool in the middle is rounding these values.
[image]

olisnr, 2024-09-17:
yes, the tool is TCL :) `w="tcleval([string map {{$} {}} [**ev** {16*$nw}]])" `...

StefanSchippers, 2024-09-17:
Ah ok, then replace `ev` with `expr`. `ev` rounds the result to 4 significant digits, it is normally used for printing final results. For data processing better to avoid any rounding.

olisnr, 2024-09-17:
super. i tried also to do an new function simply without the rounding:
```
## NEWNEWNEW evaluate expression extended precision. if expression has errors or does not evaluate return expression as is
proc eve {args} {
  set i [join $args]
  if {![catch {uplevel #0 expr $i} res]} {
    return $res
  } else {
    # puts stderr "proc ev: $res"
    return $args
  }
}
```

7 digits gives me 1mm big MOSFETs.

thanks a lot!

closed 2024-09-17 by olisnr

olisnr, 2024-09-17:
what do You think, would it make sense to write LVS-netlists with `.cdl` extension?

olisnr, 2024-09-17:
yes, i did such a thing via an command launcher.sym:

```
descr="write LVS netlist"
tclcommand="
	xschem set netlist_type spice
	set lvs_ignore 1
	set lvs_netlist 1
	set spiceprefix 0

	xschem netlist [xschem get current_dirname]/[xschem get current_name].cdl
"
```

i think, its strange that the file it generates is in the symulation directory. should the `current_dirname` not be the dirname, where the schematics sit?

StefanSchippers, 2024-09-17:
The command above will write a file `xxxx.sch.cdl` in the same place where `xxxx.sch` is present.
the same command is:
`xschem netlist [xschem get schname].cdl`

The xxxx.cdl in simulation directory is probably an older file?

olisnr, 2024-09-18:
not exactly. it goes to the simulation folder:

https://github.com/user-attachments/assets/83879b8a-54d7-47a8-b785-2a474b7bd931

StefanSchippers, 2024-09-18:
The problem is that you probably have this option set:
```
#### if this is set to '1' netlists and simulations will go into a simulation/ folder
#### inside the directory containing the top level schematic.
#### if this is set to '2' netlists and simulations will go into a simulation/[schname]/ folder
#### inside the directory containing the top level schematic.
#### Default: not set (0).
set local_netlist_dir 1
```
If this is the case the path you specify will be overridden and be `dir_of_schematic/simulation`
so add this before netlisting:
`set local_netlist_dir 0`

olisnr, 2024-09-18:
thanks, then i can do it like this (without specify a full paht):
```
tclcommand="
	xschem set netlist_type spice
	set lvs_ignore 1
	set lvs_netlist 1
	set spiceprefix 0
	set last_local_netlist_dir $local_netlist_dir
	set local_netlist_dir 0
	xschem netlist [xschem get current_name].cdl
	set local_netlist_dir $last_local_netlist_dir
"
```

StefanSchippers, 2024-09-18:
If you don't set a path (something with '/' characters) xschem assumes you want to keep the netlisting directory as is and only change the netlist filename. 
However you can change the xschem netlist command to:
`xschem netlist ./[xschem get current_name].cdl`
or
`xschem netlist [pwd]/[xschem get current_name].cdl`
if you want to generate the cdl netlist in current directory.