StefanSchippers/xschem #237: XSCHEM Configuration Error : TCL not founf

Ikram-rs22, 2024-10-03:
I'm completely new to Xschem and linux. I've install ubuntu in a virtual box. I've installed all the necessary libraries mentioned in the xschem manual but whenever I try to configure xschem, it says the following. But I have tcl8.6.15 installed. Please help me, I'm a new user
[image]

StefanSchippers, 2024-10-03:
You need to install the tcl libraries:
tcl8.6-dev
and
tk8.6-dev

See the complete list of required packages [here](https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html).

Ikram-rs22, 2024-10-11:
Sorry for the late response. Thanks a lot sir. For your co-operation

closed 2024-10-16 by StefanSchippers