StefanSchippers/xschem #238: cph() in XSchem built-in graph

tomscii, 2024-10-16:
Hi Stefan,

First of all, I am hugely grateful for your continued work on XSchem! I have started to use it for basically all my (hobbyist) analog circuit modeling work (with ngspice as the simulator backend), and it works great!

I have a small question/request regarding the built-in graphing capabilities of XSchem. When calculating an AC response (bode plot) I would like to use the ngspice `cph()` (continuous phase) function to obtain nice phase graphs. However, I only found `ph()` made available by XSchem. (Magnitude is nicely taken care of by `db20()`, no issues there!)

Here is a demonstration of the difference, displaying the results of the same network simulation. The first displayed within XSchem using `ph()` and the second via the native gui of ngspice using `cph()`. Is there a way I could use (or emulate) `cph()` from within XSchem to get a nicer phase graph there? I checked in the respective manuals (XSchem as well as ngspice) but I have not managed to find a way... maybe you have a tip of how I could export/save(?) the result of the `cph()` function from the ngspice workspace along with everything else? I tried but it did not seem to work. Simplest for me, of course, would be if XSchem offered `cph()` just as it offers `ph()` today...

Thank you so much in advance and happy to answer any questions! I'm running the latest git head of XSchem compiled locally, so I'm also happy to test a patch or anything similar, should that be of any help.

XSchem with `ph()`
[image]

ngspice native gui with `cph()`
[image]

tomscii, 2024-10-16:
Probably tangential: ngspice will also show the graph with degrees, by prepending the line `set units = degrees` to .control - not sure if it has any effect on the raw file.

StefanSchippers, 2024-10-16:
Yes I came to the same piece of ngspice code. 
The cx_cph() function calculates the phase of a complex number using the usual atan2(re(x), im(x)) function but adds or subtracts 2π to make the vector continuous to previous value.

The bottom line is: you cannot calculate this number by only looking at the single complex number, you need to look at the previous one. I Will do the same in xschem. will take a bit of time to avoid mistakes or side effects.

StefanSchippers, 2024-10-16:
> Probably tangential: ngspice will also show the graph with degrees, by prepending the line `set units = degrees` to .control - not sure if it has any effect on the raw file.

This for sure has not ettect on the raw rils since it stores complex numbers as real and imaginary part.
The return value for phase is transformed according to the `units` variable setting:

somewhere in the ngspice code:

#define radtodeg(c) (cx_degrees ? ((c) * (180 / M_PI)) : (c))

StefanSchippers, 2024-10-16:
Ok I have added operator cph()
if you pass a phase waveform, like `ph(out)` to `cph()` you get a continuous waveform:
write this expression (including double quotes) in graph dialog box: `"ph(out) cph()"`
see the different phase waveforms below. **Please test**. I have done limited validation so far.
Thank you!

[image]

tomscii, 2024-10-16:
Works perfectly!
[image]

I also checked the diff you just committed, to me it looks like a perfect port of the ngspice function, so I am pretty sure this is as good as it gets!

Thank you so much for this!

closed 2024-10-16 by tomscii