StefanSchippers/xschem #239: crash at updating plots while Xyce is still writing .raw file

olisnr, 2024-10-24:
when i simulate with Xyce, and the plots are updated while the simulations are still running, or if the simulator is killed and then the plots are updated, Xschem often crashes. in the command line it writes things like:

```
...
extra_rawfile() read: /ALL/Xschem/OTA33.xyce/simulation/OTA33_test_xyce.raw/ALL/Xschem/OTA33.xyce/simulation/OTA33_test_xyce_dc_zoom_mostt.spice.raw not found or no "tran" analysis
raw_read(): failed to open file /ALL/Xschem/OTA33.xyce/simulation/OTA33_test_xyce.raw/ALL/Xschem/OTA33.xyce/simulation/OTA33_test_xyce_dc_zoom_mostt.spice.raw for reading
extra_rawfile() read: /ALL/Xschem/OTA33.xyce/simulation/OTA33_test_xyce.raw/ALL/Xschem/OTA33.xyce/simulation/OTA33_test_xyce_dc_zoom_mostt.spice.raw not found or no "tran" analysis
read_dataset(): WAARNING: malformed raw file, aborting
free_rawfile(): clearing data
free_rawfile(): clearing data
free_rawfile(): no raw file to clear
raw_read(): no useful data found
extra_rawfile() read: /ALL/Xschem/OTA33.xyce/simulation/OTA33_test_xyce_dc_mostt.spice.raw not found or no "dc" analysis
read_dataset(): WAARNING: malformed raw file, aborting
free_rawfile(): clearing data
EMERGENCY SAVE DIR: /tmp/xschem_emergencysave_OTA33_test_xyce_ccdbcgbbgb

FATAL: signal 11
while editing: OTA33_test_xyce
```

olisnr, 2024-10-24:
and then i think the coloring of the `simulation` button on top right is a bit strange. when i start many simulations via TCL 

```
        foreach corner $corners { 
            foreach sim $simulations {
                xschem set netlist_name ${schematic}_${sim}_mos${corner}.spice
                set id_${sim}_${corner} [simulate ]
            }
        }
```

then the color changes direct to red, and after a while to green, while many simulations are still running (i can see it in the processlist window from menue `simulation::list running sub-processes` ). and after all the simulations are done without errors, it changes again to red.

my simulator command is:
`Xyce "$N" -plugin ./Xyce_Plugin_PSP103_VA.so -r "$N.raw" -l "$N.log"`

StefanSchippers, 2024-10-24:
I think I will remove all the coloring on these buttons. 
There are too many use cases where this can fail.
FIxing one will break the other.

StefanSchippers, 2024-10-24:
Can you please send the incomplete raw file that makes xschem crash? it will be easy to load, debug and add the necessary sanity checks.
Thank you

olisnr, 2024-10-24:
[test.zip](https://github.com/user-attachments/files/17510564/test.zip)

it seems Xschem only crashes if both .raw files are in the simulation folder.

https://github.com/user-attachments/assets/d5925902-75b5-49ee-8d7e-691e62ca7368

olisnr, 2024-10-24:
> I think I will remove all the coloring on these buttons...

thats also ok, then the users can do it them self... i would use:
`[xschem get top_path].menubar.simulate configure -background red`
or is there a better way?

StefanSchippers, 2024-10-24:
Ok last commit should fix the crash.
Incomplete raw files written by Xyce contain no number of points (it is in the first header of the file, No. Points: xxx), the xxx number is left blank and written in the file only when simulation finished, so that number of steps is known.

In cases when a file is corrupt xschem deletes all raw files currently loaded. Some other parts of the code where not aware of this were and still using the (deleted) raw data. Now fixed.

StefanSchippers, 2024-10-24:
It should be a good idea to file a bug request to the Xycev developers. 
If you terminate Xyce with a TERM signal (the default kill <process_id> command) Xyce should update the `No.Points` with the steps written so far in the file when intercepting the TERM signal.

If you killed the program with the -9 (KILL) signal (kill -9 <process_id>) then it's not a Xyce fault. This terminates the program immediately, no way to complete the `No. Points` line.

olisnr, 2024-10-24:
perfekt! thanks a lot

closed 2024-10-24 by olisnr

StefanSchippers, 2024-10-24:
> and then i think the coloring of the `simulation` button on top right is a bit strange. when i start many simulations via TCL
>
> ```
>         foreach corner $corners {
>             foreach sim [...]

The coloring of the buttons where done assuming simulation is run by pressing the button. When simulation starts button turns to orange and pressing again the button kills the running simulation and starts over again. Orange means "running", while green and red are mapped to the exit status of the process (green =ok, red=ko).

However running multiple simulation in parallel from a script (usually on different netlist files) is always possible, so button colors are less important.

olisnr, 2024-10-24:
yes its difficult to have use full colors in both cases, and if You mix per script started simulations with pressing buttons, then its not so clear what the result should be.

one idea could be:
if an simulation is started, now matter how, the button shows simulating, and it does this as long as there are running sub-processes.
if one or more sub-processes are running, pressing the simulation button ends all.
if all sub-processes are ended, the state of the last one is shown. or it goes simply to the default color back.

if You start many simulations per script, You still want to know when all have ended, or be able to end all. now i do all this via the process-list dialog, but per button would more elegant.

olisnr, 2024-10-25:
is there a periodic timer event in Xschem, that could be used for updating things?

olisnr, 2024-10-25:
is there a Xschem symbol for .ic inital voltage of a net?

StefanSchippers, 2024-10-25:
There is no ic symbol, but adding one is easy.

Can you try to place the attached `ic.sym` in devices/ and use it ? if it works for you I will add into mainline xschem devices/

You can also do some artwork if you prefer a different look. if it looks nice send it back i will use that one.

`ic.sym`:
```
v {xschem version=3.4.6RC file_version=1.2
*
* This file is part of XSCHEM,
* a schematic capture and Spice/Vhdl/Verilog netlisting tool for circuit
* simulation.
* Copyright (C) 1998-2023 Stefan Frederik Schippers
*
* This program is free software; you can redistribute it and/or modify
* it under the terms of the GNU General Public License as published by
* the Free Software Foundation; either version 2 of the License, or
* (at your option) any later version.
*
* This program is distributed in the hope that it will be useful,
* but WITHOUT ANY WARRANTY; without even the implied warranty of
* MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
* GNU General Public License for more details.
*
* You should have received a copy of the GNU General Public License
* along with this program; if not, write to the Free Software
* Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA
}
G {}
K {type=ic
template="name=ic1 value=0"
format=".IC V(@#0:net_name\\\\)=@value"
}
V {}
S {}
E {}
L 4 -20 -20 0 0 {}
B 5 -1.25 -1.25 1.25 1.25 {name=p dir=in}
T {IC=@value} -20 -20 2 0 0.3 0.3 {layer=4}
```

olisnr, 2024-10-25:
i will try. if You like it, You could also the FET symbols i did, that place the bulk near the source, so that You can wire the gates strait to many FETs like in the textbook schematics. 

[image]

StefanSchippers, 2024-10-25:
In sky130 we have something similar. 3 terminal mosfets with body set via `body=...` attribute. see image.
This allows to select more transistors and change the body connection for all without wiring anything.
[image]

olisnr, 2024-10-25:
this is like my X2 symbol for the 3-pin NMOS. it also uses the `body=...` but if You want to connect the body with something other then a power label, as example with the source, the label solution is not very handy. thus the 4 pin symbol with the alternative placed bulk

StefanSchippers, 2024-10-25:
yes i agree having both is the best option.

olisnr, 2024-10-25:
im happy with Your IC symbol.

olisnr, 2024-10-25:
i saw, that the processlist dialog also schows the open editors with netlists or attributs, so, its not so easy to change the colors from the number of child-processes :)