StefanSchippers/xschem #240: change in mouse-wheel behavior after tclcommand "xschem annotate_op "

olisnr, 2024-10-24:
in the newer versions of Xschem i found, that sometimes the mouse-wheel changes its function. as example, after running a `tclcommand` with 
`xschem annotate_op ${netlist_dir}/${schematic}_mos${mos_corner}.raw
`
while before the wheel changed the diagram, after it changes the main-window

https://github.com/user-attachments/assets/8c37124f-621d-4395-99b0-ae682c9024e4

olisnr, 2024-10-24:
and then the zoom is also a bit strange. the clipping could be improved. now it remembers at the cathode-ray oscilloscopes without pre-trigger :)

https://github.com/user-attachments/assets/cb756894-1a39-457a-9154-a809edb39789

olisnr, 2024-11-28:
this is still the same... i think it has something to do with cinnamon.

StefanSchippers, 2024-11-28:
I also had **a one time** problem like yours. When entering the graph the mouse pointer did not change to '+' and did not zoom the graph but the window instead. 

But I need to trigger this issue again. and this seems very difficult.

StefanSchippers, 2024-11-28:
One case when such event occurs is if you press 'm' or 'c' outside the graph with nothing selected to move or copy something (but nothing is selected so nothing happens). the UI enters move or copy mode and you can't zoom the graphs. Pressing Esc (which terminates all pending commands) fixes the issue. 

This is clearly a bug. If nothing is selected 'm' or 'c' must do nothing.

olisnr, 2024-11-29:
i dont really like that the zooming is switched from window to graph by having the mouse-pointer inside the graph. is it possible to change it to use a `modifier-key` to zoom the or `f` the selected graph? normally, i have the schematics zoomed in, to have a big view of the graph. im with the pointer with high change over a graph. then i want to zoom out, and use the middle wheel, or the `f-key` and instantly regret not first placed the pointer outside of a graph...

StefanSchippers, 2024-11-29:
This is a good point. I also was planning to do it. Having the graphs moving while working on the schematics is annoying.

I can add an option to do that. I will not change the default as some users may complain.

olisnr, 2024-11-29:
what do You think about the cliping? its more noticeable in Xyce because the time-steps are adaptive:
[image]
it this difficult to improve?

StefanSchippers, 2024-11-29:
> what do You think about the cliping? its more noticeable in Xyce because the time-steps are adaptive: 

In general xschem does not interpolate or extrapolate points to make the waveforms exactly fit the graph area. 
in the image above the first waveform point is the first one found in the raw file that is inside the x axis range. Similarly the last waveform point is the last one in the raw file that is inside the axis range. 

This shows that on that zoom level the simulation density of points is too low. You should reduce the simulator timestep (by increasing the time precision).

StefanSchippers, 2024-11-29:
> i dont really like that the zooming is switched from window to graph by having the mouse-pointer inside the graph. is it possible to change it to use a `modifier-key` to zoom the or `f` the selected [...]

the following variable is added in xschemrc:
```
###########################################################################
#### USE CTRL MODIFIER TO OPERATE ON GRAPHS WITH MOUSE & KEYBOARD
###########################################################################
#### if enabled forces to hold Control key pressed to operate on graphs
#### to prevent "graph event stealing to schematic"
#### Default: 0 (not set)
# set graph_use_ctrl_key 1
```

if that graph_use_ctrl_key variable is set to 1 you must hold the COntrol key down to do anything on graphs, otherwise the key/mouse events are directed to the schematic.

If unset everything works as used to.

olisnr, 2024-11-29:
Yeseses!! much better! Thanks!!!

about the plots, i think You should not think about interpolation or so. You could simply plot all points inside a window PLUS one point before and one after... i think the graphical APi should handle the clipping. or is this wrong?

the thing is: its not interesting to make more datapoints then nesesary: it costs time and electrical work. the nice thing at good time domain simulations is, that they adapt the sample-rate. i can also show You plots over temperature. i have only the points -55 -50 0 20 40 125. zomming looks bad. thats not necessary. or the other example is: .dc simulations over 5 corners, only the MOSFETs, not mention the resistors or the BTj that multiply the possibility even more.

its interesting to simulate whit the minimum of points, there is no real reason not to plot a line to a point outside the graph window.

[image]

what do You think?

olisnr, 2024-11-29:
[image]

this is a sweep over a voltage source, that supplies a shunt-regulator over 100Ω:
```
                \".step temp list -55 -50 0 20 40 125\\n\"\\
                \".dc vdd -.5 6 0.2 \\n\" \\
                \".print dc format=raw v(*) i(*)\"\\
```

StefanSchippers, 2024-11-30:
I have committed a fix to show full x-range of waves without clipping to existing simuation points. This done by plotting one extra point (if existing)  before and after the graph range.

This needs some testing. Can you please see if it works?

olisnr, 2024-11-30:
Yes perfekt!!! exactly like expected!

now i have to Xschems open. two different schematics & simulations. one is working with zooming the graphs via the new nice cntr-modifier, and opens the graph-dialog via double-Lclick. the other dont zooms the graph even with the cntr-modifier, but only the schematics, and the double-Lclick opens a Text-window:
[image]

i have not started Xschem over a terminal. is there some other thing i could try to find out, why this two Xschems behave so different?

StefanSchippers, 2024-11-30:
This is strange. I have started more xschem instances and can't see the issue.
Ctrl-Double Lclick the graph brings the above window (that is the graph attributes, without the dialog box presenting them in graphical format) if the graph is selected or if you click very close to the graph borders.

Are you using 2 separate xschem instances? or are you opening a second window/tab with the same xschem process?

StefanSchippers, 2024-11-30:
I have pushed another fix, in some cases if waveforms are not loaded the graph dialog box does not show up. Update xschem and test your fail example one more time. 
Thank you

olisnr, 2024-11-30:
i always open Xschem from a file-browser like nemo or nautilus via double-click. i think the problem is connected to the strange zoom-behavior You also saw 1-time. but what speaks against is, pressing `m` or `c` without selection, didnt change anything in this case.

olisnr, 2024-11-30:
https://github.com/user-attachments/assets/76388e32-0275-4225-ad1b-a5db90631ecb

olisnr, 2024-11-30:
that was the old Xschem version. i only wanted to show You the effect before closing and reinstalling Xschem...

StefanSchippers, 2024-11-30:
The behaviour in above video is correct, since the graph you are double clicking is selected (the border is solid grey line, not dashed) If you click outside the graphs on an empty area you unselect everything and the graph dialog box should appear when double clicking the graph (with control if enabled)

olisnr, 2024-11-30:
[amps_plot_problem.xyce.zip](https://github.com/user-attachments/files/17965153/amps_plot_problem.xyce.zip)
open `amps_plot_problem.xyce/hsens10_3.5mA_15dB_test_xyce.sch` and try double-Lclick a loaded plot.

olisnr, 2024-11-30:
ah ok, i missed this. now i have to ctnr also for the dialog :)

StefanSchippers, 2024-11-30:
Canyou send also the xschemrc you are using? you have a lot of options enabled so i test the same condition.

olisnr, 2024-11-30:
sure [xschemrc.zip](https://github.com/user-attachments/files/17965201/xschemrc.zip) but may be Your last changes did also fix the original zoom-problems i had anyway... 

i will try a bit and tell You if i see another graph-zoom problem if it dares to show up :)

StefanSchippers, 2024-11-30:
ok, thank you.

olisnr, 2024-11-30:
i love the new way of graph zooming

closed 2024-11-30 by olisnr

olisnr, 2024-11-30:
1 thing that seems not to work is the cntr-f over a graph to auto-zoom the hole graph. may be there is a line code missing? or is this intended?

StefanSchippers, 2024-11-30:
On my system it works. may be another key combination Cinnamon is stealing?

https://github.com/user-attachments/assets/f1007400-ccaf-4d48-b67b-6066094bc1c4

StefanSchippers, 2024-11-30:
I found this in Cinnamon. Is this true also if mouse is over an application window? this should be done only when hovering over a free desktop area...
[image]

olisnr, 2024-12-01:
cntr-f is normally for searching. cinnamon dont steal this. its used in many different SWs. the search-dialog opens also in Xschem. and the zooming the full graph is also working on my other cinnamon computer. 
but on this one i use now its not... i will install Xschem again, and tell You the result...

olisnr, 2024-12-01:
after reinstalling its working! thanks a lot for all Your work!