StefanSchippers/xschem #241: New feature suggestion: import SVG images

itrendafilov, 2024-10-26:
I really like the ability to document everything using xschem. I currently import PNG files, but they are quite bulky and colors don't match the color scheme of the editor.

I've tried doing a vectorizing an image with custom polygon over It but it is very time consuming. If you have time, would you add the ability to import SVG files?

StefanSchippers, 2024-10-26:
Hi, Thank you for this report. This is a nice idea. I need to search if there is some library that does this, before reinventing the wheel. 

You can always import s svg image into xschem by using a `filter` attribute:

in the image rectangle attributes:
```
file=/.../image.svg
filter="gm convert - png:-"
```
but this is just a svg to bitmap conversion, so it is not a solution.

StefanSchippers, 2024-10-26:
Ok I did some research. 
The `libsrvg` library from Gnome is the way to go to load a svg image and render it to a cairo surface (the object used by xschem to visualize images).

I will do that since I like the idea. But will take a bit of time to complete.

- It adds another library dependence so I need to add the necessary checks to the configure system.
- xschem should use this library if available, but should build (with no direct svg loading) if the library is not installed on the system.
- Then I need to add the code to load svg into xschem images in addition to the existing png and jpg formats.
- do a lot of testing

StefanSchippers, 2024-10-27:
I have rejected the idea of using `libsrvg` since this library is too big and introduces a ton of GNOME dependencies. I don't like the GNOME bulky stuff, so i came up with a different solution.

You need a converter from svg to png. This is used only for rasterizing the image into xschem.
The latest xschem commit does store the original svg file in the `image_data` attribute so you will not lose resolution and for vector SVGs the schematic size will be small. The best svg-->raster converter is `rsvg-convert`, from package `librsvg2-bin`.
You can get it with `sudo apt get librsvg2-bin`.

https://github.com/user-attachments/assets/06e6920b-b943-4245-8813-e0d7a587386b

StefanSchippers, 2024-10-27:
I have further improved svg loading. Now the tools-> insert image also mentions svg.

in xschemrc you can set a svg_to_png filter. see xschem in the src directory:
```
###########################################################################
#### IMPORT SVG IMAGES
###########################################################################
#### this is a filter command that processes svg files and writes png data
#### listed in my personal order of preference.
#### filter must read stdin and write to stdout.
set svg_to_png {rsvg-convert}
# set svg_to_png {inkscape --pipe --export-type=png --export-filename=-}
# set svg_to_png {convert - png:-}
```
uncomment one such filter if it is available on the system.

When inserting a svg image xschem will automatically add the filter attribute so you immediately see the image.

https://github.com/user-attachments/assets/99d89fa4-609a-4a94-82d3-4b19d92f93ae

closed 2024-11-07 by StefanSchippers