StefanSchippers/xschem #242: Managing symbols with multiple schematic views

fedeinthemix, 2024-10-31:
Hi,

Thanks for your great xschem! More than an issue, I have a feature request,.. should you find it useful as well.

I watched your video on [multiple schematic views](https://xschem.sourceforge.io/stefan/xschem_man/video_tutorials/multiple_schematics_bound_to_symbol.mp4). It is quite useful. However, when working on large designs with several dozens of sub-blocks, it becomes difficult to make sure that each symbol is configured to use the right schematic. It would be useful to have a "view manager" where you can see the list of all symbols, and edit the schematic that they are supposed to use all in one place. Some commercial tools use a "config" view associated with the top testbench schematic for that.

At work we make heavy use of this functionality, for example for top level verification where we check one-subsystem at a time for speed reasons. For this we make use of empty or simplified schematic viewws. This allows us to avoid having multiple top level schematics and avoid all the problems that they involve.

Best Regadrs

StefanSchippers, 2024-10-31:
Hi, Thank you for your post. I will try to come up with something user friendly for this.

StefanSchippers, 2024-11-05:
In your usage case do you plan to set **one** specific schematic "view" for a symbol or are you planning to use different schematic "views" for a symbol that is instantiated multiple times in a design?

I have worked a lot on memory chips, and these have multiple placements of identical blocks (arrays, row drivers, column decoders, sense amplifiers).
In full chip spice simulations we used to simulate at full transistor level only a small portion of these blocks, while using simplfied "views" for inactive blocks. These simplified views usually provide only the parasitic loading effects and few other things (like driving some outputs to the correct levels and so on).
In this scenario a symbol like "sector.sym" will probably have a "sector.sch"default implementation and a "sector_empty.sch" simplified view. (there could be more views if needed).

A placement of sector.sym with no "schematic" attribute will use the default sector.sch, while instances with a "schematic=sector_empty.sch" (or schematic=libname/sector_empty.sch, depending on how the search path is configured) will use the simplified view. This makes sector.sym polymorphic as multiple implementations may coexist in the same design.

I am asking about your use case to prioritize your specific needs.

fedeinthemix, 2024-11-06:
We always use the same view for all instances of a symbol throughout the hierarchy.

Concretely let say that we have symbols A and B. Let's also assume that symbol A has two views: 
- schematic.sch which includes instances of the B symbol, and
- simplified.sch which doesn't.

In the case where for A we use schematic.sch then all B (inside and outside of A and any other symbol) use the same specified view. In the other case where for A we use simplified.sch there are only instance of B outside of A and all those use the same specified view.

Similarly for a deeper hierarchy. From you description, it sounds like you are using a similar flow.

Thank you very much for looking into this!