StefanSchippers/xschem #244: an other crash while loading .raw file from mpirun Xyce

olisnr, 2024-10-31:
[test.zip](https://github.com/user-attachments/files/17594024/test.zip)

StefanSchippers, 2024-10-31:
The crash is confirmed.
I see the raw file has the **plotname:** line which is 226844 characters long.  A very long name for a plot. 
The rest of the file header is completely corrupt:

Title: ** sch_path: /ALL/Xschem/BG.xyce/OTA33_test_mc_xyce.Xsch
Date: Thu Oct 31 11:31:57 2024
Plotname: Step Analysis: Step 1 of 3 params:  name = X1:X12:XM10:SG1.....< 226k chars...>
Fla40: real
No. Variables: 492
No. P40nts: 40 40           40
40ri40les:

so Flags:, No.Points:, Variables: lines are corrupt. 
I will add more checks to avoid the crash, but this file can not be displayed.

Thank you for providing these corner cases.

StefanSchippers, 2024-10-31:
I have added another check so xschem does not try to go any further reading these malformed raw files.
--> no crash on these raw files.

olisnr, 2024-10-31:
great! thanks

closed 2024-10-31 by olisnr