StefanSchippers/xschem #246: Build of 3.4.6 version for windows

georgtree, 2024-11-05:
Hello!
Is it possible to get windows build of the 3.4.6 version of xschem?
Thank you in advance!

StefanSchippers, 2024-11-05:
Yes we are preparing the 3.4.6 release. Currently xschem is in release candidate phase.

georgtree, 2024-11-11:
Also, on windows, option Show grid degrades performance in the order of magnitude

StefanSchippers, 2024-11-11:
> Also, on windows, option Show grid degrades performance in the order of magnitude

Forwarding to the windows maintainer, since I don't have a windows machine.
Will keep you informed. Thanks

jlowpc, 2024-11-12:
> Also, on windows, option Show grid degrades performance in the order of magnitude

Stefan suggested to use "set big_grid_points 1" with show grid and I confirmed with that setting, the performance is comparable to turning off "draw_grid"  

>Also, how to remove image blinking in windows cairo version?
Can you provide more information on what you mean by image blinking in Windows cairo version (e.g., test case, how do I reproduce the same issue you see)?  I want to see if there are ways to improve the degradation you're seeing.   I am aware that Xschem drawing on Windows is not as smooth as Linux, but it has been tolerable for me.

georgtree, 2024-11-25:
Hello!
Is there any updates regarding windows build for 3.4.6? Thank you.

StefanSchippers, 2024-11-25:
The windows maintainer is holding the release since she is investigating graphics performace issues on the Windows OS.
This is due to the fact that some graphic functions (like clipping, multiple lines drawing etc) are missing or implemented differently than on Linux. As soon as she gets done a release will be made.

StefanSchippers, 2024-12-25:
3.4.6 release is done. 
Sources:
https://github.com/StefanSchippers/xschem/releases/tag/3.4.6
Windows binaries:
http://repo.hu/projects/xschem/releases-bin/

closed 2024-12-25 by StefanSchippers