StefanSchippers/xschem #248: Building issue

georgtree, 2024-11-11:
Hello!
I tried to build latest version of xschem, configure finishes succesfully, but when I try to make, I got this error:
```
scheduler.c: In function ‘tclvareval’:
scheduler.c:6263:3: warning: implicit declaration of function ‘va_start’ [-Wimplicit-function-declaration]
 6263 |   va_start(args, script);
      |   ^~~~~~~~
scheduler.c:6265:15: warning: implicit declaration of function ‘va_arg’ [-Wimplicit-function-declaration]
 6265 |   while( (p = va_arg(args, const char *)) ) {
      |               ^~~~~~
scheduler.c:6265:28: error: expected expression before ‘const’
 6265 |   while( (p = va_arg(args, const char *)) ) {
      |                            ^~~~~
scheduler.c:6271:3: warning: implicit declaration of function ‘va_end’ [-Wimplicit-function-declaration]
 6271 |   va_end(args);
      |   ^~~~~~
make[1]: *** [Makefile:72: scheduler.o] Error 1
make[1]: Leaving directory '/home/georgtree/Downloads/xschem-code-r3425/src'
make: *** [Makefile:2: all] Error 2
```

Maybe issue is in multiple versions of tcl installed? I have 8.6 from ubuntu repo, and 9.0 compiled manually.

Thank you in advance.

StefanSchippers, 2024-11-11:
Hi, George, 

I have added the line `#include <stdarg.h>` in the `scheduler.c `file, since it is using the `va_*` macros.
I don't know why this error didn't show up when building on my system, probably some other file or library is including this file. Anyway it shoud compile now.
Thank you for pointing this out.

StefanSchippers, 2024-11-11:
By the way I was not aware of the tcl9.0 release. Will try it out just to anticipate future breaking changes (that always happen when new versions come around...) :-). Xschem works fine with tcl8.4 and tcl8.6

georgtree, 2024-11-11:
Yes, the only issue I could see is some incompatibilities between 8.6 and 9.0 in C-API

georgtree, 2024-11-11:
Download new version and now get:
```
/usr/bin/ld: scheduler.o: in function `tclgetboolvar':
scheduler.c:(.text+0x2b8): undefined reference to `Tcl_GetBool'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:18: xschem] Error 1

```

StefanSchippers, 2024-11-12:
This seems a missing library in the linking phase. You need to add something like -ltck8.6 or -ltcl9.0 and may be the -L/path/to/lib if the libtclx.y.so is not in a standard place.

StefanSchippers, 2024-11-12:
the function I use is Tcl_GetBoolean,not Tcl_GetBool...
```
int tclgetboolvar(const char *s)
{   
  int res;
  const char *p;
  p = Tcl_GetVar(interp, s, TCL_GLOBAL_ONLY | TCL_LEAVE_ERR_MSG);
  if(!p) {
    dbg(0, "%s\n", tclresult());
    return 0;
  }   
  if(Tcl_GetBoolean(interp, p, &res) == TCL_ERROR) {
    dbg(0, "%s\n", tclresult());
    return 0;
  }     
  return res;
}     
```

georgtree, 2024-11-12:
Yes I see, I opened the mentioned file and there is Tcl_GetBoolean.
Honestly I don't have experience with tcl c interface, but maybe it is connected somehow:
https://core.tcl-lang.org/tips/doc/trunk/tip/618.md

StefanSchippers, 2024-11-12:
@georgtree ,

I have downloaded and installed the tcl-tk 9.0 source files yesterday, so I have made some small additions to allow xschem work flawlessly with this new version (it will sometime replace the long standing 8.6 in linux distributions, so it's better to be prepared).

After installing the tcl-tk package somewhwere (`/home/schippes/x/tcltk9.0` in my case) 
I set the library search path LD_LIBRARY_PATH, since /home/schippes/x/tcltk9.0 is not a standard dynamic link library search path: 
`export LD_LIBRARY_PATH=/home/schippes/x/tcltk9.0/lib`
Then after going into the xschem source files directory I ran the configure script specifying the 9.0 tcl version:

`./configure --debug --prefix=/home/schippes /arg/tcl-version=9.0 /arg/tk-version=9.0 --prefix/libs/script/tcl=/home/schippes/x/tcltk9.0 --prefix/libs/script/tk=/home/schippes/x/tcltk9.0`

and finally built xschem:

`make -j4 install`

The build was successful, 

https://github.com/user-attachments/assets/81e444b0-0961-4dec-89fc-1769a5d08f87

georgtree, 2024-11-13:
Yes it works, everything were successfully built, thank you!

closed 2024-11-13 by georgtree