StefanSchippers/xschem #249: Font issue in Tk frame

georgtree, 2024-11-14:
Hello!
How to fix this?
[image]
Cairo is installed, issue is in tk elements only, thank you!

StefanSchippers, 2024-11-14:
Sorry, what is exactly the problem?

georgtree, 2024-11-14:
Absence of antialising

StefanSchippers, 2024-11-14:
what is the tcl-tk version you are using to build xschem?

georgtree, 2024-11-14:
9.0

StefanSchippers, 2024-11-14:
On a windows machine or on a Linux system? (real or emulated, doesnt matter)

georgtree, 2024-11-14:
Fresh kubuntu 24lts, I reinstall the system today and build xschem

StefanSchippers, 2024-11-14:
ensure font packages are installed:
**fonts-freefont-ttf  fontconfig fonts-liberation2 fonts-liberation fonts-dejavu**.

You can check if the font issue is generic with tk 9.0: 
start the `wish9.0` shell, write these commands in it:
```
% button .a -text abcdefghijkABCDEFGHIJK
.a
% pack .a
```

do you see the same non aliased characters in the button?

[image]

StefanSchippers, 2024-11-14:
Also at the xschem prompt issue this command to check if xft is used. (xft = X FreeType interface)

```
% ::tk::pkgconfig get fontsystem
xft
% 
```

If you don't get 'xft' then tcl-tk is using a bitmap font system and you probably need to change some font configuration file in your home directory (in ~/.config or other places).

georgtree, 2024-11-14:
I installed all fonts, and run ::tk::pkgconfig get fontsystem, the result is x11

georgtree, 2024-11-14:
Ok i checked configure log for Tk, and found this:
checking whether to use xft... no
need to find why it doesn't detect it

closed 2024-11-14 by georgtree

StefanSchippers, 2024-11-14:
Yes I was writing this: install **libxft-dev**
Good!