StefanSchippers/xschem #251: Add constants qualificator in graph drop-down menu

georgtree, 2024-11-21:
Hello!
I found that there is no word constants in drop-down menu here:
[image]

StefanSchippers, 2024-11-21:
I will add the constants menu.
In the mean time you can write it in the dropdown textbox menu. 
[image]

georgtree, 2024-11-25:
Hello!
This updates broke how it works previously, now constants always reads as complex, and shows strange things when try to plot. I used constants for reading data that were extracted with .meas statement as vectors in raw file.

StefanSchippers, 2024-11-25:
Do you have an example raw file to send me for inspection?
If the ngspice convention is to store complex data in the "Constants" section I must read as complex data.
If there are also "Complex" sections with real data I must handle this.

georgtree, 2024-11-25:
Previously it was shown like it:
[image]

StefanSchippers, 2024-11-25:
Thank you, your raw file explains the issue. ngspice can save both real and complex "constants" sections:

```
Title: Constant values
Date: Tue Oct  8 15:02:40 UTC 2024
Command: ngspice-43+, Build Tue Oct  8 15:02:40 UTC 2024
Plotname: constants
Flags: complex
No. Variables: 12
No. Points: 1

```

```
Title: Constant values
Date: Wed Feb 14 13:10:01 UTC 2024
Plotname: constants
Flags: real
No. Variables: 5
No. Points: 5

```

Ok, understood, will fix that. Will look at the **Flags** line instead of assuming a data type.

StefanSchippers, 2024-11-25:
Ok above fix is done. If you have a chance please test.

georgtree, 2024-11-25:
Just checked, it works well, thank you!

closed 2024-11-25 by georgtree