StefanSchippers/xschem #252: question about loading .op data from Xyce into Xschems annotation

olisnr, 2024-11-27:
Xyce has a different handling of multiple simulations then ngspice, and writes only a log-file in the .op simulation. 
if i simulate something with ngspice, i can have the .op data in the raw file and annotate the schematics with it. but i didnt found out how to do it with Xyce simulations. 
i tried a 1 point .dc simulation, but it didnt work. 
what is the way?

StefanSchippers, 2024-11-28:
in Xyce the way to store OP data is the following:
```
.op
.print dc format=raw file=test_op.raw v(*) i(*)

```
This saves a 1-point DC sweep.
I have pushed a fix in xschem: if a OP-annotation is requested ad a "Operating Point" section is not found (because Xyce was used with the above command lines) then go ahead, try to load a "DC Transfer Characteristic" section and use the 1st point for annotating the schematic.

Thsi will allow to annotate correctly the schematic for ngspice and Xyce OP raw data.

Thank you for pointing out this issue.

olisnr, 2024-11-28:
thanks a lot!
i was reading the Xyce documentation and didnt found this. why do You know this?

olisnr, 2024-11-28:
i tried it, but it didnt annotate. what could go wrong? should i send You an example?

StefanSchippers, 2024-11-28:
Here is the example I tried. If you have a simple exampel that does not work please send it to me so I can look at it.

https://github.com/user-attachments/assets/39d2680b-cc42-466b-9e29-69329f481787

StefanSchippers, 2024-11-28:
> i tried it, but it didnt annotate. what could go wrong? should i send You an example?

Just numerous trial and error...

Also this paragraph of the Xyce User Guide gives a hint...
**7.2.3.
OP Analysis
Xyce also supports .OP analysis statements. In Xyce, consider .OP as a shorthand for a single-step DC
sweep, in which all the default operating point values are used.**

olisnr, 2024-11-28:
i was this also reading, but i think we should ask the Xyce if they dont want to update the doc...

my example needs works with: `Xyce -plugin $env(PDK_ROOT)/$env(PDK)/libs.tech/xyce/plugins/Xyce_Plugin_PSP103_VA.so "$N"  -l "$n.log"`
but the plugin isnt needed because i desabled the iHP amps...

the OP data are: 
```
amps.xyce.op_ano/simulation/hsens10_test_xyce_mostt.log
amps.xyce.op_ano/simulation/hsens10_test_xyce_mostt.spice.raw
```

[amps.xyce.op_ano.zip](https://github.com/user-attachments/files/17951552/amps.xyce.op_ano.zip)

StefanSchippers, 2024-11-28:
I tried the example, but to me it seems working...

https://github.com/user-attachments/assets/9821bb43-9b69-454e-b29a-2e30826db8fc

Are you doing something in a different way?

olisnr, 2024-11-28:
i do exactly the same. i will download the newest software and try again...

StefanSchippers, 2024-11-28:
Yes, please go with the newest. The option for looking into a DC section after failing to find an OP section (as ngspice does) is very recent in xschem.

StefanSchippers, 2024-11-28:
If it does still not work there is probably some hidden corner case to be figured out.

olisnr, 2024-11-28:
its working with the newst version! thanks a lot and sorry for not downloading the newest variant first!

closed 2024-11-28 by olisnr