StefanSchippers/xschem #255: question to the cntr-modifier and graph-dialog

olisnr, 2024-12-03:
the cntr-modifier is used to deselect selected things in Lclick, but also to trigger launchers. while modifies selecting via rectangle to a drag mode.
may be it would be clearer if 
cntr-Lclick selects in drag-mode
cntr-rect selects in drag-mode
Lclick selects in move-mode
rect selects in move-mode
shift Lclick toggles selection
shift rect toggles selection
l launsches the current selected launcher... or something similar
what do You think?

the graph dialog is sometimes a bit strange, as example if i use `i()` without `"`:

[image]

would it be possible to include the `"`, so that its not needed anymore (also for calculations)?

StefanSchippers, 2024-12-03:
Deselection is done with Alt+LClick, so it is different from Ctrl-LClick.
The only problem with this is on Macs as they don't have the Alt key (they have the Command key).
But in xschem the Windows key can also be used instead of the Alt key, and to my knowledge the Mac Command key is mapped the same as the  PC Windows key.

olisnr, 2024-12-03:
for multi-line there is the `\`.

olisnr, 2024-12-03:
> Deselection is done with Alt+LClick, so it is different from Ctrl-LClick. The only problem with this is on Macs as they don't have the Alt key (they have the Command key). But in xschem the Windows [...]

yes `cntr` is the way of KLayout to deselect... :) but cntr Lclick would really be nice for the drag-selection...

StefanSchippers, 2024-12-05:
It is now possible to enter wave expressions on one line with spaces without requiring double quotes.
quotes are allowed and are necessary if you want to span a wave expression on multiple lines (this is useful if expression is very long). The backslash for line continuation was too difficult to implement considering that xschem applies color tagging to wave names/expressions in the graph dialog box, so for multiple lines use "..."

Please do some tests.

**There is one side effect:**
In some graphs I used to write signal lists like:
RST DATA CLK
This does not work any more, you have to put these on different lines:
RST
DATA
CLK

Thank you

olisnr, 2024-12-05:
is it possible, that the change makes TCL code different?
i used to have lines like:
`diff0;i(vsingle_all)%[rawfilestart]_temp_mosff.spice.raw
`
in `""` and now its not working anymore. i will try to get it work, but probably You know already how it actual works :)

olisnr, 2024-12-05:
the `""` free graph-dialog is working perfect for me. much nicer. and i think You also improved the text-dialog much. i can undo and redo, thats new, or not?

thanks a lot!

closed 2024-12-05 by olisnr