StefanSchippers/xschem #257: After generate netlist file name in the tab changes V3.4.6

Dunkelwind, 2024-12-13:
start:
[image]
then I press Netlist
result:
[image]
saving switch back to top.sch

StefanSchippers, 2024-12-13:
Thank you for reporting this bug.
I can not repeat this problem, so you probably need to prepare a test schematic and attach it here.
In the mean time i will do tests on the numerous examples i have.

Dunkelwind, 2024-12-13:
Sorry I'cant repeat this bug. But this bug appears sometimes by doing schematic design session.
As far as I remember, just clicking on Netlist leads to the status modified.
I think that the saving of the modified SR3OR.sch was correctly done in top.sch, SR3OR.sch was not changed when saving (as one would assume).

StefanSchippers, 2024-12-13:
Yes the error is probably triggered by some combination of events. It is probably happening when working with more than one tab. Will do some tests.

StefanSchippers, 2024-12-13:
If it happens again and you remember what you did just before let me know, so I can restrict my tests. Thanks

StefanSchippers, 2024-12-13:
Just for information, does the netlisting of top.sch take a long time? (I mean, is it a very big circuit?) So I will do tests on a similar condition.

Dunkelwind, 2024-12-13:
no, complete spice-netlist size is 33.889 Bytes.

Dunkelwind, 2024-12-13:
I have created a new sub circuit. It's from an another, save as, edit design.
"A" to make sym.
correct some bugs in the sub circuit schematic
new tab
open in new tab top.sch
insert these new symbol
connect symbol with wires/ports
click Netlist
top.sch now SR3OR.sch*
save
top.sch in the tab

StefanSchippers, 2024-12-13:
Thank you, I will go through the steps and see...

StefanSchippers, 2024-12-14:
is SR3OR.sch the schematic of the symbol (SR3OR.sym) that you inserted in top.sch?

To correct the errors in the symbol sub schematic did you descend into it from top.sch or did you fix it in the first tab?

StefanSchippers, 2024-12-14:
I am trying al weird combinations following the instructions you gave above, can not see any broken tab name...

StefanSchippers, 2024-12-14:
please can you record a desktop session to show the error? while recording also tell the keys or mouse actions you are doing. 
I think I can replicate this error only if I follow your actions to the atomic level.

StefanSchippers, 2024-12-14:
Another question: are you using cinnamon desktop environment?
Cinnamon DE so far is the one causing most of unreplicable errors, due to design flaws in the desktop, like event stealing, global grabs and such.

Dunkelwind, 2024-12-14:
_is SR3OR.sch the schematic of the symbol (SR3OR.sym) that you inserted in top.sch?_
is top/Register8/SR3OR
_To correct the errors in the symbol sub schematic did you descend into it from top.sch or did you fix it in the first tab?_
fix it in the first tab
_I am trying all weird combinations following the instructions you gave above, can not see any broken tab name..._
I am trying too, without problems. No idea what exactly is causing the problem.
_Another question: are you using cinnamon desktop environment?_
xfce4
Mint 5.15.0-126-generic #136~20.04.1-Ubuntu SMP Thu Nov 14 16:38:05 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

StefanSchippers, 2024-12-14:
I am trying to do this. Can you please look at the video and tell if I am doing something different? I don't see the problem.

https://github.com/user-attachments/assets/d629c7d1-09c2-484c-be4d-80b5d4877d2a

Dunkelwind, 2024-12-14:
At the end of the video you have to click on Netlist. This may reveal the problem.
But as I've already written, I can't always reproduce it.

StefanSchippers, 2024-12-14:
Ok, thank you for now.

StefanSchippers, 2024-12-14:
Do you have some specific setting in the xschemrc file?

Dunkelwind, 2024-12-15:
I'm use xschem direct from /home/dunkelwind/xschem-src/src/. 
And thus also the rc file available there.
Here the video:

https://github.com/user-attachments/assets/e1018045-583d-41ab-8339-6a881d978657

StefanSchippers, 2024-12-15:
Thank you for the video. Will do some other tests

StefanSchippers, 2024-12-15:
When you press the Netlist button the ERC (errors) window appears. What is the block with errors?

StefanSchippers, 2024-12-15:
I am very frustrated. 
I think this error is a unique combination of factors that can not be reproduced in any other environment.

StefanSchippers, 2024-12-16:
Another question: if you exactly do again the actions shown in the video does the error happen again? or only sometimes?

StefanSchippers, 2024-12-16:
Question 3: when the tab changes name does the window title also change name?
[image]

StefanSchippers, 2024-12-16:
I was able to trigger the error. Now I can work for a fix. 

https://github.com/user-attachments/assets/c34dddee-e708-48ca-bf30-874d223e1659

Can you enable the option shown in the video and let me know if it does the same error?

StefanSchippers, 2024-12-16:
The issue above is now fixed. Let me know if this happen again on updated xschem.
Thank you.

Dunkelwind, 2024-12-16:
Tnx for the bug-fixing,
Yes, if Auto join/trim is selected, the error occurs.s.

StefanSchippers, 2024-12-25:
Closing for now. If issue happens again fell free to reopen...

closed 2024-12-25 by StefanSchippers