StefanSchippers/xschem #258: Create raw file from arbitrary data

georgtree, 2024-12-19:
Hello!
There is a way to create raw file with command `xschem raw new name type sweepvar start end step`, also we can add new vectors to existing raw file, or modify existent  ones. But, is there  a way to create raw file from arbitrary data, for example something like:
`xschem raw new rawName sweepvarName $sweepvarData vector1Name $vector1Data ...` ?
It would be great to see this ability in xschem, thank you.

StefanSchippers, 2024-12-19:
There are various ways.
One is to read data formatted as an ascii table:

```
X     Y     Z
0     0     0
1     1     1
2     4     8
3     9    27
4    16    64
5    25   125     
```
The first line provide the vector names, the first column is the x axis, 
all other columns are data to be plotted. Lines beginning with '#' are ignored, 
double empty lines start a new dataset (a new line, disjoint from previous)
single empty lines are ignored.
If above lines are saved in a file named 'table1' the command to read the table in xschem is:

`xschem raw table_read table1`

you can then plot the data as usual:

[image]

The second method is to create data in memory:

`xschem raw new test test1 x -10 10 0.01`

Creates a new plot test1, from in-memory data 'test' (you can have many plots by changing their names)
with X-axis range form -10 to 10 and 0.01 step. 
A 2001 element 'x' vector is created. {-10, -9.99, .... , 0, ..., 9.99, 10}

`xschem raw add y { x x * -1 * exp() }`

Adds a 'y' variable equal to exp(-x^2).

[image]

This command: 

`for {set i 900} {$i < 1100} {incr i} { xschem raw set y $i 1 }`

Pokes into the data and changes part of the gaussian to a box.

[image]

You can also do something like this:

```
xschem raw new test test1 x 0 3 1 ;# create a 4 item x vector (0,1,2,3)
xschem raw add y ;# add an y variable (all zero, (0,0,0,0)
# fill x and y with new data
for {set i 0} {$i < 4} { incr i} {
  xschem raw set x $i [expr $i/10.]
  xschem raw set y $i [expr exp($i)]
}
```
[image]

the attached example loads a table with measured power [kW] vs speed [km/h] for some different car types.

[image]

[power_vs_speed_cars.zip](https://github.com/user-attachments/files/18201755/power_vs_speed_cars.zip)

georgtree, 2024-12-26:
Got it, thank you very much :)

closed 2024-12-26 by georgtree