StefanSchippers/xschem #259: Possible bug in wiring

ananthchellappa, 2024-12-22:
Is it possible the code could be improved? It's adding solder dots incorrectly and in some cases, it hasn't registered the connection for one wire..
[image]

StefanSchippers, 2024-12-22:
This is not a bug
partially overlapped wires will show dots, see figure.
[image]

partially overlapped wires can be merged with the '&' command (`Tools -> Join.Trim wires`)
this operation can be made the default with `Options-> Auto Join/Trim wires` or by setting 
`set autotrim_wires 1` in the xschemrc file. If AutoTrim is enabled behavior is pretty much the same as Cadence.

The "unmerged wires" behavior is sometimes desired, so xschem does not automatically merge wires if we want to keep all the pieces separate. In below image I can select the piece of wire, delete it and insert a current meter. Of course there are multiple ways to do this task, but this is just a user option.
[image]

StefanSchippers, 2024-12-23:
But yes, I see some connecting dots added whie doing Undo.This is a bug. Will look into that.

StefanSchippers, 2024-12-23:
The bug you found is hopefully fixed. (connecting dots appearing on unrelated wires when doing an Undo)
If you have a chance to do some tests let me know if you find other inconsistencies related to connecting dots (not related to overlapping wires, which is known and not a bug (use auto Trim/Join for that).

ananthchellappa, 2024-12-23:
I'll check Stefan. I'm looking into the docs.

Is it easy to update Xschem using an apt install command?

Thank you,
Ananth

ananthchellappa, 2024-12-23:
I used git clone and then configure/make and I am not able to reproduce this issue, suggesting that it's fixed.

However, it may be worth nothing that I see this in the xterm from which the local (not /usr/local/bin/xschem) was launched:

```
ttuser@ubuntu-desktop:~/Downloads/xschem-src$ src/xschem 
Sourcing /usr/local/share/xschem/xschemrc init file
Sourcing /home/ttuser/.xschem/xschemrc init file
can't read "graph_use_ctrl_key": no such variable
xschem [~/Downloads/xschem-src] can't read "graph_use_ctrl_key": no such variable
can't read "graph_use_ctrl_key": no such variable
can't read "graph_use_ctrl_key": no such variable
can't read "graph_use_ctrl_key": no such variable
can't read "graph_use_ctrl_key": no such variable

can't read "graph_use_ctrl_key": no such variable
can't read "select_touch": no such variable
can't read "graph_use_ctrl_key": no such variable
can't read "select_touch": no such variable
can't read "graph_use_ctrl_key": no such variable
```

olisnr, 2024-12-24:
> I'll check Stefan. I'm looking into the docs.
> 
> Is it easy to update Xschem using an apt install command?
> 
> Thank you, Ananth

apt is very slow and behind. i would suggest You to download a zip from https://github.com/StefanSchippers/xschem/archive/refs/heads/master.zip
and install from source (in a terminal @ */xschem-master).
./cofigure
make -j7
sudo make install

http://repo.hu/projects/xschem/xschem_man/install_xschem.html tells You what You need to install before. if You did it once, its easy

StefanSchippers, 2024-12-24:
As @olisnr pointed out it takes months to get a new release packaged into linux distributions.
To incorporate these very new changes either take the master tar.gz file or clone the xschem github repo. [Building compiling and installing](https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html) is not dfficult, once you have the required libraries and dependencies installed.

ananthchellappa, 2024-12-29:
No issue in latest release. Thanks for fixing.

Will request that wire command stay in "command" mode till ESC is pressed - so user an draw a multiple-segment-wire without clicking W each time for a new wire command. Cadence-style routing will come eventually 😊

closed 2024-12-29 by ananthchellappa

olisnr, 2024-12-30:
@ananthchellappa try `double left click` to end the track. then You dont need to `w` again :)