StefanSchippers/xschem #260: Feature to add in graph section: y-value displayed with matched color for each waveform near the intersection with the b" cursor

grai365, 2024-12-25:
If possible it would be very useful to have the Y-values displayed with the same color as each of the waveforms, for multiple waveform graphs, near the intersection with the "b" cursor. For example if I run Ibias for 3 temperatures I would like to see the cursor sampled values for each temp, on the graph, right now we can only see the back-annotated  value corresponding to the first temperature

StefanSchippers, 2024-12-25:
It is easy to do. Add one waveform per dataset as shown below.

[image]

The general format for a wave is:

`alias_name; name%run#`

the `alias_name;` is optional, so you can just have in the example above:
```
diffout%0
diffout%1
diffout%2
```
The numbers referto  the spice run number.

StefanSchippers, 2024-12-25:
It is also possible to display only a single waveform and have the value displayed for that specific wave. Point the mouse close to a waveform and press 't' (or ctrl-t if graph_use_ctrl_key is enabled in xschemrc). ALl other waveforms will be switched off. See video.

https://github.com/user-attachments/assets/9d2bfc66-e79c-4ca0-989f-54f415a23103

StefanSchippers, 2024-12-25:
There is also a "live measurement" tool. Pres 'm' inside a graph to toggle the measurement tooltip.

https://github.com/user-attachments/assets/f7405327-330f-4594-89d0-5c704c996847

closed 2025-01-17 by StefanSchippers