StefanSchippers/xschem #261: Possible issue with move (unintentional copy)

ananthchellappa, 2024-12-28:
Only the connected instance of the diode is behaving this way.

[image]

pasting the testcase.sch :

v {xschem version=3.4.6 file_version=1.2}
G {}
K {}
V {}
S {}
E {}
T {Goal is to use RMB click and
drag to zoom.} -160 -140 0 0 0.4 0.4 {}
N 40 -10 40 30 {lab=#net1}
N 40 30 90 30 {lab=#net1}
N 90 30 90 50 {lab=#net1}
C {ammeter.sym} 40 -40 0 0 {name=Vmeas savecurrent=true spice_ignore=0}
C {diode.sym} 90 80 0 0 {name=D1 model=D1N914 area=1}
C {diode.sym} 90 80 0 0 {name=D2 model=D1N914 area=1}
C {diode.sym} 270 130 0 0 {name=D3 model=D1N914 area=1}
C {diode.sym} -60 130 0 0 {name=D4 model=D1N914 area=1}
C {diode.sym} -120 -40 0 0 {name=D5 model=D1N914 area=1}
C {diode.sym} 230 -70 0 0 {name=D6 model=D1N914 area=1}
C {diode.sym} -200 90 0 0 {name=D7 model=D1N914 area=1}
C {diode.sym} 310 20 0 0 {name=D8 model=D1N914 area=1}

StefanSchippers, 2024-12-28:
The problem is that you have D1 and D2 overlapped. You move D1 and D2 remains in its position
you can also see from the testbench:
C {diode.sym} 90 80 0 0 {name=D1 model=D1N914 area=1}
C {diode.sym} 90 80 0 0 {name=D2 model=D1N914 area=1}
the 4 numbers (X Y rotation flip)  are the same.

StefanSchippers, 2024-12-28:
It is also shown in the info windows if you try to generate a netlist:
Error: overlapped instance found: D2(diode.sym) -> D1

ananthchellappa, 2024-12-28:
Sorry about that :) Thanks for taking a look so soon.

How difficult is it to do a "Schematic check" utility like Cadence has? Most of the required code might already exist since the netlist generator is able to flag such an issue.

Things I am interested in:

1. Unconnected pins (inputs especially)
2. Solder dot on crossover
3. Shorted outputs

closed 2024-12-28 by ananthchellappa