StefanSchippers/xschem #266: Menu Bar Drop-Down Does Not Dismiss or Update on Interaction

ananthchellappa, 2024-12-30:
(Could I get an **estimate of effort**? That will help me look into recruiting others to fix this)

When interacting with the menu bar in the application, the following issues are observed:

Drop-Down Not Dismissing: After opening a menu drop-down (e.g., by clicking "File"), clicking the same menu item again does not dismiss the drop-down, contrary to expected behavior.

Drop-Down Not Updating on Hover: If a drop-down menu is open (e.g., the "File" menu), hovering over adjacent menu items (e.g., "Edit") does not update the drop-down to reflect the hovered menu item. This behavior prevents seamless navigation between menu items.

Steps to Reproduce:

1. Launch the application and open the menu bar.
2. Click a menu item (e.g., "File") to reveal its drop-down.
3. Click the same menu item again; note that the drop-down remains open.
4. With the drop-down still open, hover over another menu item (e.g., "Edit") and observe that the drop-down does not change to reflect the hovered item.

Expected Behavior:

1. Clicking the same menu item should dismiss the currently open drop-down.
2. Hovering over adjacent menu items while a drop-down is open should update the drop-down to reflect the hovered item seamlessly.

Actual Behavior:

1. Clicking the same menu item does not dismiss the drop-down.
2. Hovering over adjacent menu items does not update the open drop-down.

Video shows the contrast with NEdit-NG - which has the desired behaviour. 

[image]

StefanSchippers, 2024-12-30:
> Hovering over adjacent menu items while a drop-down is open should update the drop-down to reflect the hovered item seamlessly.

this is now done. Need to figure out how to close the menu drop down  by pressing again the menu title. Will do in a separate commit.

StefanSchippers, 2024-12-30:
> Clicking the same menu item should dismiss the currently open drop-down.

latest commit fixes this too.

https://github.com/user-attachments/assets/f087b38b-bc23-45d6-936e-eb0368a96d37

StefanSchippers, 2024-12-30:
This is almost impossible to do. 
Huge work, probably unportable because we have to modify the intricate details of the Tk library and minimal return on investment.

StefanSchippers, 2024-12-31:
I have completely rewritten the menu system using the more modern menubar vs the menubutton approach (see tk menu manpage).

The only thing that can't be done is closing the pull down menu by clicking again on the menu. 
Once I get this to work i will add another commit.
sub cascade menus wil now be opened when hovering the mouse on them.

ananthchellappa, 2024-12-31:
Thank you Stefan! I was going to take a look, since chatGPT gave this script that demonstrates the required behaviour, suggesting that TCL/Tk can do it.

`package require Tk

# Create a menu with a cascade
menu .menubar
menu .menubar.file
.menubar add cascade -label "File" -menu .menubar.file
.menubar.file add cascade -label "Open Recent" -menu .menubar.file.openrecent
menu .menubar.file.openrecent
.menubar.file.openrecent add command -label "File1"
.menubar.file.openrecent add command -label "File2"

# Set the menu
. configure -menu .menubar

# Bind hover events
proc bind_hover_to_menu {menu} {
    set count [$menu index end]
    for {set i 0} {$i <= $count} {incr i} {
        if {[$menu type $i] eq "cascade"} {
            set submenu [$menu entrycget $i -menu]
            bind $menu <Enter> [list $menu postcascade $i]
            bind_hover_to_menu $submenu
        }
    }
}

# Attach bindings
bind_hover_to_menu .menubar

# Run the application
pack [label .label -text "Hover over menus"]
`

ananthchellappa, 2024-12-31:
When I build, I see that commit b23b162f2 doesn't have the expand-on-hover functionality for sub-menus (such as Open Recent), but maybe that is still WIP. I see that dismiss on 2nd click does work..

StefanSchippers, 2024-12-31:
this is latest behavior: the only thing that does not work is dismiss on 2nd click

https://github.com/user-attachments/assets/407b9316-647d-4315-a4a7-a134b81ef9ed

ananthchellappa, 2024-12-31:
OK, if I do a make clean, then configure and then make I get the same behaviour shown here.

Re non-dismissal on 2nd click, the other thing I notice is that, with the menu displayed, if you press ESC to dismiss it, it cannot be redisplayed by clicking on the item name again without first moving the mouse out of that menu name.

That is, move mouse over File and click LMB to display File menu. Now, press ESC to dismiss the menu. Now, clicking on File again doesn't display the menu. You have to move the menu out of the File area (just the text of "File") and come back if you want to display the File menu again.

In case it helps..

ananthchellappa, 2024-12-31:
OK, I said that the mouse has to move "off" the menu header (Eg File) before clicking it will display the menu again (after pressing ESC to dismiss it).

Not true - it is enough if the mouse moves even a little. If the mouse has moved, then another click displays the menu

ananthchellappa, 2024-12-31:
I notice that xschem.tcl contains a lot of functions and also the structure of the GUI - menu items, etc.

May I suggest another approach? (coming from the world of Cadence where there are a literally hundreds of companies using the product and each has it's own GUI customizations)

We split the menu components and functionality into separate files so that the project is more maintainable.

There could be a core.tcl with core non-negotiable items like File, Edit, etc, accompanied by a core_func.tcl - containing supporting functions.  Maybe we can use better names - core_gui.tcl, core_gui_func.tcl, etc.

Same for simulation.tcl, etc. 

Having stuff in a file related to a particular function might make things more manageable as we try to scale this project to incorporate the full functionality of Cadence. There is no reason it cannot be done..

StefanSchippers, 2024-12-31:
It is easy to customize menus and toolbar buttons. See the example files in the src/ directory:
`add_custom_menu.tcl` and `add_custom_button.tcl`:
see here the **PDK** (this one I have enabled in my xschemrc to switch process PDKs), **Test** and smiling face toolbar button.

[image]

The reason I keep one big tcl file for the gui is one of the reasons I wrote xschem when escaping from Cadence. Cadence is too big and slow. Xschem must start up in less than 200ms on a slow machine. Reading many files is a performance killer, expecially on machines with no SSD or other solid state storage.

ananthchellappa, 2025-01-02:
Any hope of getting the dismiss-on-2nd-click to work Stefan? Just curious

StefanSchippers, 2025-01-02:
I think there is no easy solution. The tk library does not provide a way to do this - just click away from the menu or press Esc.
It can be probably done by changing some internal callback function, but since these are not in the public API it is almost guaranteed to fail when a new version of the library is relased some years later when no-one remembers why the hack was done.

StefanSchippers, 2025-01-03:
There is a dirty hack implemented in xschem to make a click on an open menu item close the menu.
You need to move the mouse pointer by at least 1 pixel to reopen it with a click.
Please test. If it works please close the issue.

StefanSchippers, 2025-01-03:
Sorry , reverted the change. This thing is damn impossible to do without some nasty collateral effects/

StefanSchippers, 2025-01-04:
Committed another change. This one seems to work with no unwanted side effects.

A click on an open menu item will close the menu.
You need to move the mouse pointer by at least 1 pixel to reopen it with a click.
Please test. If it works please close the issue.

StefanSchippers, 2025-01-04:
> You need to move the mouse pointer by at least 1 pixel to reopen it with a click.
with latest commit this is no more needed. Seems menus are now working as you asked.

ananthchellappa, 2025-01-04:
It is working Stefan. Thank you!

Could you describe the changes? It will help others later to know 

what was learnt from the experience
What would have helped get it right in the beginning.

Finally, the knowledge gained is the most valuable thing to share.

One thing I noticed again, not for the first time. I do a git pull, to get the latest, then, make clean, .configure (may be unnecessary, make) then, if I run src/xschem, I don't see the fix. But, if I do a sudo make install, then I see the fix. Do we understand why?

I'll submit a new issue for the help menu dialog to show build information. Otherwise, there is no way to know if someone is using the correct version.

Again, grazie mille,
Ananth

closed 2025-01-04 by ananthchellappa

StefanSchippers, 2025-01-04:
This is the cryptic addition (in `xschem.tcl`) that fixes the menu dismiss on click:

## allow to unpost menu entries when clicking a posted menu
```
if { [info tclversion] >= 8.6 } {
  bind Menu <Button> {
     if { [%W cget -type] eq "menubar" && [info exists tk::Priv(menuActivated)]} {
       %W activate none
     } else {
       %W activate @%x,%y
     }
     tk::MenuButtonDown %W
  }
}
```
The tk lib default binding is:
```
bind Menu <Button> {
    tk::MenuButtonDown %W
}
```

So any time a button is pressed deactivate the menu if it is already activated or activate it if it is not (Activated means the menu entry has input focus).
The default <button> binding is then executed. This `tk::MenuButtonDown `procedure does the actual work of displaying the menu drop down or clearing the drop down.

This is actually NOT an xschem fix, it is more a Tk library fix. The above can be copy pasted to any Tk application.

StefanSchippers, 2025-01-04:
> One thing I noticed again, not for the first time. I do a git pull, to get the latest, then, make clean, .configure (may be unnecessary, make) then, if I run src/xschem, I don't see the fix. But, if I do a sudo make install, then I see the fix. Do we understand why?

To run a local xschem you *MUST* cd into the src/ directory and then execute xschem from there:

```
cd xschem-src
./configure
make -j4
cd src
./xschem
```

StefanSchippers, 2025-01-04:
If you want to run a local xschem build and run from outside src/ do something like this:
`/home/schippes/xschem-src/src/xschem --tcl 'set XSCHEM_SHAREDIR /home/schippes/xschem-src/src'`

The XSCHEM_SHAREDIR tells xschem where to look for its runtime needed files.
You can set this in your ~/.xschem/xschemrc (but remember to unset it if you are doing a regular full installation)