StefanSchippers/xschem #268: Inconsistent Behavior of Persistent Wire Command Between Shortcut and Menu Usage

ananthchellappa, 2025-01-02:
The behavior of the wire creation command is inconsistent when the "Persistent wire/line place command" option is enabled, depending on whether the command is accessed via the 'w' shortcut or the menu buttons.

Shortcut ('w'):

Double-clicking the left mouse button (LMB) or pressing ESC ends the current multi-segment wire creation but keeps the program in wire-create mode. The next LMB click starts a new wire.

Menu Button:

Double-clicking LMB or pressing ESC ends the multi-segment wire creation and also exits wire-create mode entirely.
This inconsistency can confuse users and reduce workflow efficiency.

Proposed Behavior:

Pressing ESC should end the wire creation process and also exit the wire-create mode.
Double-clicking LMB should end the currently created wire but retain wire-create mode to allow immediate creation of another wire.

StefanSchippers, 2025-01-02:
You are right. 
This inconsistency is now fixed

When starting a wire from the menu a following mouse click will start the wire placement.

While in 'persistent mode' and inserting a wire segment a first Esc clears the segment that follows the mouse.
A second Esc clears the wire insertion command altogether.

While in 'persistent mode' and inserting a wire segment a double click confirms the segment.
wire insertion is still active, so next mouse button click starts a new wire.
An Esc clears the wire insertion command altogether.

ananthchellappa, 2025-01-02:
Verified fix. Thank you Stefan.

I still feel ESC (vs double-click) should finish and end, but that's just one point of view..

closed 2025-01-02 by ananthchellappa