StefanSchippers/xschem #270: 3.4.6 version for windows issue

georgtree, 2025-01-02:
Hello!
Just tested windows build of 3.4.6, and I encounter next issue:
performance improved, that's great, but at certain zooming, it degrades a lot, and if I zoom further, it returns back to good performance.
So, there are intermediate levels of zoom at which the performance degrades a lot.

StefanSchippers, 2025-01-02:
xschem uses spatial hash hables when it is reasonable to do so.
if doing a full zoom xschem iterates over all objects and displays all of them that fit into the screen.
if zooming in but still lot of objects to show xschem still iterates over all objects but skips drawing off screen items.
if zooming into a small area a different algorithm is used. a spatial table is created and all objects are hashed based on their position. Spatial hash table allows to quickly iterate over the only elements that are within a box.
This is much faster if the schematic is very large and viewport is small,  but adds overhead if we must show a large portion of it, so in this case it is not used.

So there is a zoom level in between where you see a change in frames per second.
I think the problem you see is due to general performance problems on windows that are still unadressed.

georgtree, 2025-01-02:
Ah, also the moving of elements degrades significantly in comparison to previous windows build (3.4.5) at all zoom levels

georgtree, 2025-01-02:
yeah I see, for now just continue to use Linux version on WSL, performance is super there

closed 2025-01-02 by georgtree