StefanSchippers/xschem #271: Can't go to upper level of hierarchy in read only files

georgtree, 2025-01-02:
hello!
If I open file that is read-only, modify it and then try to Ctrl+E, it throws:
[image]
Previously it asks about the save, I just clicked no and easily go upward.

StefanSchippers, 2025-01-02:
Also verify you **don't** have this setting in the `xschemrc` file:
`set no_ask_save `1
this option is discouraged for various reasons.

StefanSchippers, 2025-01-02:
I have found the piece of code handling the failed save. 
if the save was not doable due to some protection on files the "go up" routine did a return so not allowing to go up in hierarchy. I have changed this behavior now. If a write is not possible just issue a warning and go ahead anyway. Thank you for the report!.

closed 2025-01-02 by georgtree