StefanSchippers/xschem #273: Running a script through Tools > Execute TCL Command

ananthchellappa, 2025-01-02:
If we have a script of commands, is this the correct approach to run it? I know that a new instance of Xschem can be launched with the --script option. This is about an already running instance of Xschem. If we need to run a script, what is the correct procedure? 

Thanks and apologies if this is covered in the manual. I looked, but didn't find it.

ananthchellappa, 2025-01-03:
Thank you so much Stefan. Any chance we can get this information included in the user guide at : https://xschem.sourceforge.io/stefan/xschem_man/xschem_man.pdf

One thing I am interested in is a log file (like Cadence's CDS.log) (I know we already get a log from dbg print statements, but that doesn't help with automation) where EVERY interaction is captured. If you have any ideas/suggestions on how to implement that, I can look into volunteering for that. It should benefit everyone.

StefanSchippers, 2025-01-03:
You can change some dbg() functions into a different function like **xschem_log()** and direct the output to some log file (**xschem.log**)

StefanSchippers, 2025-01-03:
xschem has the 
FILE *open_tmpfile(char *prefix, char *suffix, char **filename)
function that opens a unique filename in XSCHEM_TMP_DIR, returns an open FILE pointer you can write into and returns also the name of the file in filename.
This is needed so if you have multiple xschem instances they will have their own private log file.
so:
```
char filename[PATH_MAX];
FILE *log_fd;
log_fd = open_tmpfile("xschem_", ".log", &filename);

```
will create a unique file `xschem_sdfku7jdAg.log` in `XSCHEM_TMP_DIR` (tipically in `/tmp`)

StefanSchippers, 2025-01-03:
this man page has been added:
https://xschem.sourceforge.io/stefan/xschem_man/scripts.html

closed 2025-01-03 by StefanSchippers