StefanSchippers/xschem #274: Inconsistent behavior of selection (left to right Vs right to left)

ananthchellappa, 2025-01-03:
Observations:

1. Wires are always included in the selection the same way, but not instances.
2. If the first corner (click and then drag) is on the right of the opposite corner, instances are included as soon as the selection rectangle crosses the instance.
3. However, if going from left to right, the entire instance needs to be enclosed.

Suggested update :

Make this the same as Cadence:

1. There will be an option variable that says whether an object (anything wire, instance, etc) needs to be crossed or enclosed by the selection rectangle.
2. Only the "final" rectangle will be used to include objects (currently, even if an object is crossed by the selection rectangle, which rectangle, when "final" does not enclose OR cross the object, it gets included)

[image]

StefanSchippers, 2025-01-03:
A tcl variable controls this:
```
#### if set select objects if touched by selection rectangle while dragging to the left
#### Default: enabled (1)
# set select_touch 0
```

You might also consider this:
```
#### if set show selected elements while dragging the selection rectangle.
#### once selected these can not be unselected by retracting the selection rectangle
#### if not set show selected items at end of drag. Default: enabled (1)
# set incremental_select 0
```

ananthchellappa, 2025-01-04:
Thank you Stefan. Good to know it's by design. Is it possible to include this in the PDF manual? That is probably the first place someone will look.. Sorry if it's already there. I looked, but didn't find it.

closed 2025-01-04 by ananthchellappa