StefanSchippers/xschem #275: Feature addition to the waveform graph: adding horizontal cursors

areda0, 2025-01-06:
I couldn't find anything about horizontal cursors in the manual. It's nice to have two horizontal cursors similar to the vertical "a" and "b" cursors.

StefanSchippers, 2025-01-06:
There are no horizontal cursors yet. 
In the mean time, you can add arbitrary horizontal traces with a simple trick. write:
```
" 10.0 "
```
in the list of signals in the graph dialog box and you will see a horizontal line at value 10.0.  Add a space inside the quotes so it will be clear it is a constant expression (value 10.0) and not a spice node named 10.0.

By pressing 'ctrl-m' inside a graph a meter tooltip appears when you hold the mouse pointer still inside a graph.
[image]

areda0, 2025-01-06:
Very nice. Is it possible to modify the line type? For example, I would like to change the "low threshold" horizontal line to a dashed line, with a smaller width, without affecting other plots in the same graph.

StefanSchippers, 2025-01-06:
Currently not, but I will add something for this.

StefanSchippers, 2025-01-06:
Creating H cursors turned out to be much simpler than I thought. More work will be needed to put in place all the mouse bindings to move / position the cursors. I will implement two cursors to allow showing Y-differences.
Will look like this:
[image]

areda0, 2025-01-06:
Perfect. Thanks @StefanSchippers .

areda0, 2025-01-08:
@StefanSchippers 
I couldn't get it to work as expected. 
Shift+A is "enabling show netlist window"
[image]

shift+B is showing this text window: 
[image]

I'm pointing inside the graph. Tried with and without "set graph_use_ctrl_key 1". Am I missing something?

areda0, 2025-01-08:
@StefanSchippers 
working as expected.