StefanSchippers/xschem #277: Non-default infix and orthogonal wiring mode

ananthchellappa, 2025-01-17:
I just mention these updates in a fork of the project so they may be considered for merging. This is the fork : https://github.com/TheSUPERCD/xschem

One option to give users is the non-infix mode (that is, mouse position at command entry is NOT the first point). Some users prefer to click explicitly to start a wire. This has been implemented only for wire mode.

Orthogonal mode : after the first point is entered, the angle that the line from the first point to current mouse position is used to decide whether the next segment is drawn horizontally or vertically. When the mouse is clicked, only the first segment is finalized and the next segment is drawn in rubber band mode.

This is an interim solution. Ideally, we want what Cadence gives us - which takes into account whether the start point is on a pin and likewise for the end point and intelligently draws three or more segments. That will obviously take time.

The other thing is the command mode indication on the status bar (minor bug). We do want to change this to a prompt message. That is pending.

[image]

xschemrc : updates for defaults for new variables
xschem.tcl : new menu entries, message on wire mode on status bar
callback.tcl : update for support for non-infix use-mode and implementation of orthogonal routing. Thank you Chayan!

StefanSchippers, 2025-01-17:
Ortoghonal wiring can be activated by pressing the space bar while placing the wire.
The decision whether to use oblique or manhattan paths is under the user control. The tool should not decide this.
This was one of the most hated ''features'' of Cadence and our team immediately asked for disabling that forever, together with any ''autorouting'' feature that always routed wires in the most damaging way.

[image]

StefanSchippers, 2025-01-17:
Also, after pressing a wire or line command, pressing 'h' or 'v' will enforce horizontal or vertical movement.

[image]

ananthchellappa, 2025-01-18:
Hi Stefan,
       Really appreciate that. Can you give us insight into how to implement it globally?

That is, infix (as implemented in Cadence) applies to "everything". If set, then, regardless the command, if a coordinate input is required, the mouse position at command entry is taken as the first coordinate.

Therefore, a true implementation of infix (that is, option infix_interface 0, which is what **I** want 😊) will NOT take mouse position at command entry (but wait for a click) for a command such as SHIFT-M (move with wires attached) - as an example..

If we can do that, that would be great, but I understand if there are higher priority items.

StefanSchippers, 2025-01-18:
I don't know if this can be done globally.

Many actions on existing objects require to first select the object(s), then decide what to do.

After selecting some objects you can go in the Edit menu and click one of the 'Move Objects' entries.
Xschem waits for a mouse click to start the move operation. This is basically the Non Infix notation. It can be added easily.

I don't think there is a one-liner catch all change that does this thing globally

The intuitive interface if enabled does not even require to enter bind keys

- mouse click on an object and drag away --> move
- Shift + mouse click on an object and drag away --> duplicate object
- Ctrl + Shift + mouse click on an object and drag away --> move adding wires
- Ctrl + mouse click on an object and drag away --> move the objects and existing wire endpoints attached to it

The complete list is [here](https://xschem.sourceforge.io/stefan/xschem_man/commands.html),

ananthchellappa, 2025-01-18:
Hi Stefan,
         The one I want (I'm sure I'm not the only one 😊) is : 

mouse click on an object and drag to STRETCH (not move)  - first on your list. Stretch means keep wires attached and redraw them as necessary.

I know this is not trivial. We want wires to stay in orthogonal mode.

Any suggestions on the coding would be very helpful. I'll look into the implementation.

There are documents on using the tool. If you can create a document describing the implementation and the code, that would help coders get involved and be productive with coding enhancements.

Grazie!

itsayellow, 2025-02-14:
I agree that having the tool try and figure out its own orthogonal wiring paths seems likely to give me wires I don't want.

Instead, I would love to have a setting that would only allow the two "manhattan" wiring modes (and omit the oblique mode).  You would still cycle through the two manhattan modes with the space bar, but there would be no oblique mode and you wouldn't start wiring in oblique mode.

OR I would even prefer with the same 3 wiring modes, except you would start wiring in the first manhattan mode.  That way oblique mode would still available.

Basically I never want oblique wires (except for 0.1% of the time) and it's always the default.

As far as lines (not wires) go, they're fine as is, when drawing lines (usually for symbols) I want all kinds of lines.

StefanSchippers, 2025-02-14:
> OR I would even prefer with the same 3 wiring modes, except you would start wiring in the first manhattan mode. That way oblique mode would still available.

I think this is a good suggestion. Which one of the two manhattan modes is preferred? **horizontal, then vertical** or **vertical, then horizontal**? I would guess the first one should be preferred.

StefanSchippers, 2025-02-14:
Changed the default orientation for wires to manhattan H-V. space bar can be used to toggle different orientations.

https://github.com/user-attachments/assets/4e660138-58cd-4764-8274-c7f1d2f88864

itsayellow, 2025-02-14:
Thanks Stefan, for me this simple change improves my wiring efficiency greatly!

closed 2025-03-09 by StefanSchippers