StefanSchippers/xschem #279: Cadence style yellow-dot cursor

ananthchellappa, 2025-01-18:
I know we can use ALT-X to toggle visibility of the cross-hair cursor, but I am looking into the implementation of the Cadence style cursor was below video demonstrates. 

Any guidance related to the code will be very helpful.

I guess one starting point would be the implementation of draw_crosshair and its calls, but I am not making much progress and Copilot isn't being very helpful.

Thanks

[image]

ananthchellappa, 2025-01-21:
I have something that approximately works. It needs help though:

1. The cursor size should be fixed, in pixels and should NOT scale with the zoom level. I see that this is an issue with the cross-hair marker as well (2nd video below). So, if that is being fixed, maybe you can help us on this one too..
2. The "restore" is obviously not working. At default zoom level, it's in a "lucky" state..
3. I just edited the draw_crosshair, but some users might like that, so it would be nice to create a new function tied to another variable for cursor.

[image]

[image]

`void draw_crosshair(int what)
{
  int sdw, sdp;
  dbg(1, "draw_crosshair(): what=%d\n", what);
  sdw = xctx->draw_window;
  sdp = xctx->draw_pixmap;

  if(!xctx->mouse_inside) return;
  xctx->draw_pixmap = 0;
  xctx->draw_window = 1;

    // Rectangle dimensions for the 2x2 cursor
    double cursor_x1 = xctx->mousex_snap - 1;
    double cursor_y1 = xctx->mousey_snap - 1;
    double cursor_x2 = xctx->mousex_snap + 1;
    double cursor_y2 = xctx->mousey_snap + 1;

  if(what != 2) {
    if(fix_broken_tiled_fill || !_unix) {
      MyXCopyArea(display, xctx->save_pixmap, xctx->window, xctx->gc[0],
           0, (int)Y_TO_SCREEN(xctx->prev_crossy) - 2 * INT_WIDTH(xctx->lw),
           xctx->xrect[0].width, 4 * INT_WIDTH(xctx->lw),
           0, (int)Y_TO_SCREEN(xctx->prev_crossy) - 2 * INT_WIDTH(xctx->lw));
  
      MyXCopyArea(display, xctx->save_pixmap, xctx->window, xctx->gc[0],
           (int)X_TO_SCREEN(xctx->prev_crossx) - 2 * INT_WIDTH(xctx->lw), 0, 
           4 * INT_WIDTH(xctx->lw), xctx->xrect[0].height,
           (int)X_TO_SCREEN(xctx->prev_crossx) - 2 * INT_WIDTH(xctx->lw), 0);
    }  else {
      // drawtempline(xctx->gctiled, NOW, X_TO_XSCHEM(xctx->areax1),
      //      xctx->prev_crossy, X_TO_XSCHEM(xctx->areax2), xctx->prev_crossy);
      // drawtempline(xctx->gctiled, NOW, xctx->prev_crossx, Y_TO_XSCHEM(xctx->areay1),
      //      xctx->prev_crossx, Y_TO_XSCHEM(xctx->areay2));
      
    }
  }
  if(what != 1) {

    // filledrect(xctx->crosshair_layer, NOW, cursor_x1, cursor_y1, cursor_x2, cursor_y2, 3, -1, -1);
    drawline(PINLAYER, NOW, cursor_x1, cursor_y1, cursor_x1, cursor_y2, 0, NULL);
    drawline(PINLAYER, NOW, cursor_x2, cursor_y1, cursor_x2, cursor_y2, 0, NULL);
    drawline(PINLAYER, NOW, cursor_x1, cursor_y1, cursor_x2, cursor_y1, 0, NULL);
    drawline(PINLAYER, NOW, cursor_x1, cursor_y2, cursor_x2, cursor_y2, 0, NULL);
  }
  // draw_selection(xctx->gc[SELLAYER], 0);
  xctx->prev_crossx = xctx->mousex_snap;
  xctx->prev_crossy = xctx->mousey_snap;

  xctx->draw_window = sdw;
  xctx->draw_pixmap = sdp;
}`

StefanSchippers, 2025-01-21:
axes and crosshairs now are always drawn as thin lines regardless of zoom.
There is an old xschem option:

```
#### enable to scale grid point size as done with lines at close zoom, default: 0
# set big_grid_points 0
```

If this is set (default: not set) axes, crosshairs and grid dots scale in width as all other lines when zooming in.

StefanSchippers, 2025-01-21:
I have added the small square -style crosshair:

[image]

StefanSchippers, 2025-01-22:
Some progress. If you hold down the Shift key while moving the mouse the yellow square jumps to the closest snap point (instance pins or wire endpoints) If you press 'W' you start a wire from there. If you release the 'W' and move the mouse keeping Shift down you see the closest point the wire will land if you press the mouse button. Pressing the space bar toggles the orientation (H-V, V-H, Oblique)

[image]

ananthchellappa, 2025-01-22:
Incredible! Ask and you shall receive. Thank you Stefan. I think Virtuoso's days are numbered 😊

For the snap cursor, the [Super Chayan fork](https://github.com/TheSUPERCD/xschem) has an implementation that currently only misses constant pixel size irrespective of zoom. **You may have suggestions on how to implement that.**

- Press w (not SHIFT w) to enter wire drawing mode
- Nearest snap point is indicated by "snap" cursor.
- Press 's' (run sim is now CTRL-R)  will start the wire by snapping to that point.
- Move mouse as needed, clicking to insert points. 
- Nearest snap points are still indicated by snap cursor.
- Press 's' at any time to end the current wire by snapping to the snap point.

Basically, same user experience you get with Cadence. After 25 years, it's hard to let go 😊 (yes, his fork has the orthogonal wiring default setting. Next step forward will be full-Cadence style routing, taking pin orientation into account. The philosophy is - yes, give user max control, but make her life easy by making the same decision she would have anyway, through good design)

[image]

StefanSchippers, 2025-01-22:
These functions (snap to closest object) must be done in a clever way otherwise it is easy to run into performance issues on very big schematics (the so called "_Windows-ization_" process)
You can not loop through all elements to look for the closest one. You need to restrict the search to "nearby" objects only, where in this context "nearby" means the part of the schematic that fits into the view rectangle you are zoomed into.
This is done by using _spatial hash tables_, where the infinite drawing area is divided into small squares and all schematic objects hashed according to the square they are into (of course all corner cases of objects that cross more squares must be handled. For example XQ1 in image below spans across 4 adjacent squares.

[image]

So I can do a search "what are the objects inside this square?" and this search must be fast and not dependent on circuit size.

The snap to closest point function does not show any lag on a circuit with half a million components/nets and CPU set to 400MHz (the lowest frequency I can use).
Keep in mind that functions bound to Motion events (cursor moved) are expensive, as they are called for every single pixel mouse movement.

https://github.com/user-attachments/assets/e124f103-c639-479a-8b42-6dfa8922a7b6

StefanSchippers, 2025-01-22:
Given the above implementation it is also easy to define an extension around the mouse to have a "magnet" effect (snap to closest object), while outside this region the cursor does not snap anywhere.

ananthchellappa, 2025-01-25:
Thank you very much. Chayan's implementation of snap cursor (diamond) uses the regular cursor and only its change of state (far as I know) to trigger the searching. So it should be more efficient..

I will file a new issue to request pulling that one in. He is in the process of writing up the minimal changes needed for that feature merge.

Next most urgent thing for me (before I start using Xschem for real design 😊) is to get fluid move/stretch/keep-connected of wire segments. One can currently do

CTRL + LMB-click-drag to select a wire
Then, CTRL+M and then click and click to "stretch" a wire segment.

I am looking to get the same UX as Cadence - namely, just click and drag without any formalities. 

Any suggestions on how to modify the code for that will be very helpful.

closed 2025-01-25 by ananthchellappa