StefanSchippers/xschem #280: Auto termination of current multi-segment wire when point entered is instance pin

ananthchellappa, 2025-01-18:
One enhancement I am looking into is to get the current wire to be terminated (same as ESC or double-click) when the point entered (by clicking) coincides with the pin of an instance.

Yes, I know one can double-click to end the wire, but it will be a welcome enhancement. The UX must be made as good as possible.

If I can get some guidance related to the code, that would be very helpful.

Once this is implemented, it will be a stepping stone to the implementation of the snap feature that Cadence has (cursor is on a nearby "snap" point and you can just press 's' to complete current wire)

Thanks.

[image]

StefanSchippers, 2025-01-18:
xschem has snapped wire placement. Use the Shift key in addition to 'w'. Wire will start / end to the nearest pin or wire endpoint.

Latest xchem also allows to simple click with Shift key down to end a wire placement landing on the nearest instance pin or wire endpoint.

As usual pressing the space bar toggles oblique, H-V and V-H constrained wire.

[image]

ananthchellappa, 2025-01-18:
Thank you Stefan! You already put in a new feature.. highly appreciated.

Let me explain the use-model I want to enable. Maybe you can suggest ways to hack the code to achieve it. Yes, I know that SHIFT-W will begin the wire by snapping to a nearby point such as pin of end of an existing wire.

What I want :

Press 'w' to entire wire draw (not SHIFT-W 😊. That would be for thick wire).
Now, (with the yellow dot cursor implemented 😊) we see which nearby point is the snap point and move the mouse if a different snap point is desired.
Press 's' to snap to that point and begin the wire.
Move the mouse to close to the destination and see where the cursor is indicating is the snap point.
Press 's' to complete the wire.

A lot of these are akin to coming back to dialup modem from high-speed internet and being surprised. Sure, we can use the tool as is to get work done. But, one could also program in assembly language. It's just more difficult. That's the reason I'm trying to enhance the UX 😊

Grazie

P.S. In the above video, the wire from pin 3 of the leftmost instance *should* start with a horizontal segment - that is how Cadence would do it - taking into account the position of the pin relative to the instance. Yes, this is not easy to code, but it adds a lot of value. So I will be looking into it.

StefanSchippers, 2025-01-18:
> P.S. In the above video, the wire from pin 3 of the leftmost instance should start with a horizontal segment - that is how Cadence would do it - taking into account the position of the pin relative [...]

The fundamental idea is that in xschem the USER decides the wire direction. Using the space bar makes it really easy to find the key and change the orientation.

This was one thing that was most hated of Cadence Virtuoso, drawing wires always in the wrong way.
the tool must not make decisions, only execute commands.

ananthchellappa, 2025-01-19:
Maybe, but no reason to let that stop you from helping me build the tool I want 😊

StefanSchippers, 2025-01-19:
Sure !!

There are some differences with Cadence.

## Design decisions

- Small. Xschem must be small, with the TCC compiler you can avoid building xschem and run directly from source files, just-in-time compile and startup time is less than [200ms on a laptop](https://www.linkedin.com/posts/activity-7250269753494163457-W1V2?utm_source=share&utm_medium=member_desktop).
- No check & save. The circuit is always in a checked state. Of course no ERC messages are printed while working on a schematic, a netlist extraction does also print / display errors and warnings. Since xschem is well coupled with the circuit topology it is possible to write logic rules for components and see the logic propagation in real time in a schematic. Without using any external simulator. This capability can be used to track net propagation through devices (a diode symbol for example can be instructed to propagate the net highlight only in one direction), mos devices can be programmed to propagate forward biased P-N paths. 

[image]

- Netlist generation must be instant for small to medium designs. (<500ms)
- It is possible to descend in hierarchy, copy with ctrl-C a piece of a circuit, go up in hierarchy and paste (ctrl-V) the circuit. Also possible to copy between xschem windows/tabs also of different xschem processes.
- Ascii file format, [documented](https://xschem.sourceforge.io/stefan/xschem_man/developer_info.html)
- A decent text font (Cadence was really horrendous at that)
- Extremely easy to add custom menu or buttons (see below video)
- A single xschem instance can be used to work on different pdks in different windows or tabs. See example below with 3 different designs from different pdks / projects.

[image]

- No EDA team needed to design new symbols, [Pcells, symbol/schematic gnerators or similar](https://xschem.sourceforge.io/stefan/xschem_man/tutorial_symbol_generators.html).

## Priority items

- For now my highest priority is to write  a tool to easily associate schematic implementations to symbols (something Cadence calls Hierarchy Editor (feature requested by @fedeinthemix) . These features are already in xschem , but a user friendy interface is needed, both at symbol level (associate a schematic view to a symbol, for all symbols in the design) or instance based (change the associated schematic view only for one or some specific instances of a symbol). This is taking significant work.

[image]
- Another high priority action is to leverage more mixed signal (specifically spice + Verilog) co-simulation. Some working example is provided but [really simple](https://www.youtube.com/watch?v=g9WPjZ_e8co)

ananthchellappa, 2025-01-19:
The Hierarchy Editor is a very powerful utility - sounds simple, but, once in place, user productivity can increase a lot. If we do get a Hierarchy Editor, I would suggest a way to enable users to "find" the cell in the HE by clicking on an instance in the schematic and pressing a bindkey. To see what I am referring to, 3:40 in this video here where I refer to it and the value of such a shortcut : https://www.youtube.com/watch?v=raTVgloGCFg&t=220s

ananthchellappa, 2025-01-19:
Do we know what would be needed to get support for veriloga? I know that is outside the scope of Xschem, because it's a simulator thing..

ananthchellappa, 2025-01-19:
I would highly recommend a "manual" for the code. Currently, the tool is the product and there is a manual for the users of the tool. If we create a document describing the code, then the users of the code would be developers who are interested in enhancing the tool. I think it would be very valuable as it would get you more helpers. Something to think about..

StefanSchippers, 2025-01-19:
Verilog-A works very well with ngspice by using the `openvaf` Verilog-A compiler. I have put above the link to a youtube video

ananthchellappa, 2025-01-19:
That's incredible. I'll take a look at it. Grazie!

closed 2025-03-09 by StefanSchippers