StefanSchippers/xschem #281: Is is possible to draw an outline rectangle?

ananthchellappa, 2025-01-20:
When I use the menu item to draw a rectangle, I get a filled rect. Is it possible to get one with outline only?

More questions, to give some background..

Is it one editor for schematic and symbol? I ask because I created a symbol from a schematic and, when I tried the delete the default rectangle in the auto-generated symbol, I was unable to - I could only delete line by line. Can we update it to auto-generate a symbol using a rectangle rather than 4 lines? 

Grazie

[image]

StefanSchippers, 2025-01-20:
Yes, click the rectangle, press 'q' and set fill=0

StefanSchippers, 2025-01-20:
> Is it one editor for schematic and symbol? 

Yes, there is no difference between schematics and symbols

> I ask because I created a symbol from a schematic and, when I tried the delete the default rectangle in the auto-generated symbol, I was unable to - I could only delete line by line. Can we update it to auto-generate a symbol using a rectangle rather than 4 lines?

This is a good idea. The symbol generation script is so old, at the time schem did not have outline rectangles or polygons. Time to update.

StefanSchippers, 2025-01-20:
`This is a good idea. The symbol generation script is so old, at the time schem did not have outline rectangles or polygons. Time to update.`

Now a single polygon object is used. It can be selected as a single entity, resized, filled, points can be added or deleted.

[image]

ananthchellappa, 2025-01-20:
Thank you :) Because, the next thing I was going to point out was the the rect cannot be resized, only moved..

So, this takes care of symbol generation. 

Can we also update the rectangle object to allow resizing by moving points and edges? 

It would be nice if the property form and getprop show the coordinates. That way, scripts can access them..

Grazie!

StefanSchippers, 2025-01-20:
You can resize rectangles of course.
- Legacy method: drag around a corner with Ctrl key pressed, then press 'm'. You can do this for 1 or 2 corners
- Intuitive interface enabled: click and drag a corner. (picking the internal part of a corner)

In the video, first 2 movements using the click and drag interface, then selecting corners with ctrl key down and pressing 'm'

[image]

closed 2025-01-21 by StefanSchippers