StefanSchippers/xschem #282: Opening the just (auto) created symbol.

ananthchellappa, 2025-01-20:
Hi,
        Is this behaving as expected? I would expect it to open the symbol that was just created, but it doesn't..

[image]

Also, I am on 6adee35b89862f42b5f7285dfc14f67c3f4e569d, where the comment says update make_sym.awk to use a polyline..  but, when I look at the symbol, it still has multiple lines..

StefanSchippers, 2025-01-20:
You have screwed up things

schematics have a .sch extension
symbols have a .sym extension

Usually xschem recognizes if what you are about to open is a symbol or a schematic, but in extreme cases set manually: Options->Netlist format -> Symbol global attrs for a symbol  view and
Options->Netlist format -> Spice netlist for a schematic view

You edit a sch file, then press 'a' to make a symbol, then you can open the symbol (.sym extension!)

[image]

ananthchellappa, 2025-01-20:
I guess I'm not following. Is this expected?

[image]

I assumed that, if one is editing a schematic and uses that menu item, then Xschem will open the symbol for that "cell". In Cadence, the ALT+2 bindkey does that.. Yes, I know there are a lot of bad things about Cadence 😊

StefanSchippers, 2025-01-20:
Ah, I see. This menu was intended to open a symbol in a new tab *after* selecting a symbol in the schematic. But I can change that, if nothing is selected open the symbol for the whole schematic.

In the mean time you can right click on the tab and select 'edit symbol'

[image]

ananthchellappa, 2025-01-20:
OK, that's nice. If ALT+2 is not used for anything, I would suggest binding it to that schematic view --> symbol and vice versa 😊

The latest pull gets me the rectangle in the symbol, so that's resolved. Thanks!

closed 2025-02-12 by StefanSchippers