StefanSchippers/xschem #283: Mouse position, instead of snapped mouse position (aka cursor) being used

ananthchellappa, 2025-01-23:
Hello Stefan,
         Re your point about how functions bound to mouse events is expensive, that's true.

Is it possible that we can get around that by using the cursor position change to trigger those "checking" functions? Let me explain.

As the video shows, though a cursor is implemented, it is not being used - the mouse pointer is still being used. You can see cases where the cursor is on an object but clicking does not select the object. So the snap grid is serving only the purpose of constraining object positions.

What I mean is, we put in the computation to only check (with every one pixel movement of the mouse) whether the cursor changed its position by snapping to a different point.
Only if it does, do we do the search (the more expensive computation) to update the snap-cursor for wiring start/end snapping.

Let me know if that could be explained better.

Grazie!

[image]

ananthchellappa, 2025-01-24:
Chayan offered to look into fixing this one. He is already using this approach to reduce computational cost in the find snap vertex search.

TheSUPERCD, 2025-01-25:
If I am allowed to add a little bit more: This issue turns into a bug if the screen-wide crosshair is enabled instead of the above-shown small-rectangular cursor. Since the object selection depends on the actual mouse coordinate (which is invisible in full-screen crosshair mode) and not on the grid-snap coordinate - the on-screen object isn't selected even when the crosshair is right on top of it. 

An easy solution for this without hampering the user's freedom can be implemented like this: https://github.com/TheSUPERCD/xschem/commit/260beffe717163602fb6023399d18d8f9440eda8

ananthchellappa, 2025-01-25:
Verified that this is fixed in CD's fork.

StefanSchippers, 2025-01-25:
This is now merged in. Thanks

In general when dragging a selection rectangle  or selecting objects it is desired that the mose / rectangle coords are not snapped to the grid.
This allows to select small objects without switching to smaller grid.
The exception is when using the crosshair instead of the mouse pointer. In this case the user does not "see" the real mouse pointer, **so it is good to use snapped coordinates**.

When the "small" crosshair is used (the square marker) then the mouse pointer is visible, so unsnapped coordinates should be used. See below gif for selecting ridicously small objects without requiring a grid change.
@TheSUPERCD I have changed your commit to the following:

       if(draw_xhair && crosshair_size == 0) {
         sel = find_closest_obj(xctx->mousex_snap, xctx->mousey_snap, 0);
       } else {
         sel = find_closest_obj(xctx->mousex, xctx->mousey, 0);
       }

where crosshair_size is set to tclgetintvar("crosshair_size") at the beginning of callback(), so the snap coordinates will be used only if the mouse pointer is hidden.

[image]

ananthchellappa, 2025-01-25:
I'm not sure that is good Stefan :) - to filter based on which cursor is being used.. In general, very small objects should not be drawn. Those are the things that lead to loss of productivity. In Cadence, there was occasionally the issue of some users drawing wires after reducing their snap grid. Now other users could not click on them without reducing their snap grid. It should be discouraged.

Is it possible to put in a variable so that users can choose if they want raw mouse coordinates to be used?

As you pointed out before, it will lead to computational inefficiency from searching based on raw coordinates.. I could be wrong..

TheSUPERCD, 2025-01-26:
Thanks a lot for the quick merge @StefanSchippers. Also, about the point brought up by @ananthchellappa regarding selecting very small objects, zooming in should make any small object big enough to be selected using the square-shaped crosshair. And in case the user wants to select a small object while zoomed-out (let's assume zooming-in poses a problem to their workflow), the user can disable the crosshair with a simple ALT+X and keep working, since a square-crosshair won't really be of much use in a highly zoomed out state. 

Plus, changing the crosshair into a square can only be done by explicitly running a TCL-command, so any new user will be confused why the select behavior suddenly changed after they read the docs and changed the grid-snap cursor's shape to match their preferences. Keeping the same behavior for every crosshair-size will eliminate that possibility.

A small suggestion from my side, speaking from the POV of a newbie to this awesome project. Also, thanks a lot for your hard work over the years to make this software possible. Going through the codebase really made me appreciate your continuous efforts towards making this great tool accessible to everyone.

closed 2025-03-09 by StefanSchippers