StefanSchippers/xschem #286: Better error message when error encountered in xschemrc

ananthchellappa, 2025-01-25:
This is the error:

`set snap_cursor size 3 # should be set snap_cursor_size 3 - missing _
`

But the message is:

[image]

Question : is there some option one can run with to get a better error message (for example which line caused the issue)?

StefanSchippers, 2025-01-25:
Added a line number in the error message. Thank you for pointing that out. I am often to lazy to add these simple but helpful things.

[image]

StefanSchippers, 2025-01-25:
Specifically comments are statements in Tcl, and statements must be separated by a newline or a ';'.
So to add a comment on the same line do:

set a 3 ;#this is a comment

ananthchellappa, 2025-01-25:
Grazie Stefan. That helps.

closed 2025-01-25 by ananthchellappa