StefanSchippers/xschem #287: Support for click-and-drag of wire immediately after entering stretch (move + stretch wires) command mode

ananthchellappa, 2025-01-25:
If, after doing CTRL + 'm', one does a click and then another click (after moving the mouse), the wire is stretched. In this condition, click and drag *is* supported. Click and drag is preferred as that is less effort on the users part, since it's usually for a small distance.

However, the click and drag is NOT supported when that is the first action attempted after CTRL + 'm'.

Is this something that can be addressed? 

[image]

StefanSchippers, 2025-01-25:
Just clck the net (or an instance) with Ctrl key down and move the mouse. (`Intuitive Click & Drag Interface` must be enabled in Options menu (it is by default). No key was used below other than Ctrl (and shift to select 2 objects).

[image]

ananthchellappa, 2025-01-25:
OK, that helps. Thanks Stefan.

closed 2025-01-25 by ananthchellappa