StefanSchippers/xschem #289: Resize wires based on direction of movement after clicking on end

ananthchellappa, 2025-01-25:
Maybe some users have gotten used to starting new wire segments by clicking and dragging on the end of a wire. It was never that way in Cadence. Click and drag on a wire vertex was always resize.

Is it possible to have a variable that gives the user the option? 

If `wire_end_drag_is_stretch` is set to 1 (default 0 for existing behavior), then, clicking and dragging on the end of a wire will not start drawing a new segment but will modify the existing wire?

StefanSchippers, 2025-01-25:
unselected wire and click and drag an endpoint --> add a new wire.
selected wire and click and drag and endpoint -->  resize wire
http://repo.hu/projects/xschem/xschem_man/commands.html

[image]

ananthchellappa, 2025-01-26:
Thank you Stefan. This can be improved 😊

Because, for the resize, we are requiring an extra step - having the select the wire. Intuitive UX is critical to productivity and satisfaction.

Could we put in a variable to give the user the option to dispense with being able to click and drag to draw wire? It is far easier to use 'w' and the 's' to snap (once that feature has been completely debugged by Chayan, I'll in a merge request).

ananthchellappa, 2025-01-26:
Since, for stretch, we have CTRL+click, maybe we can put that in for the wire-vertex as well. CTRL+click-and-drag on wire vertex to resize. 

And then a variable that requires/does-not-require the CTRL. 

That should keep all users happy.

StefanSchippers, 2025-01-26:
I can do that. 
But I still think that selecting a wire you want to stretch is far better in situations like this.
If the tool decides the wire to pick it will always be the wrong one. People have crushed the computer mouse and destroyed keyboards working with Virtuoso due to these wrong decisions.

[image]

Another situation is when 2 or 3 or more wires end in a common point. Clicking the join point to stretch it is not clear which wire we are about to stretch. Using a prior selection will make things explicit.

[image]

ananthchellappa, 2025-01-26:
All valid points Stefan. Now that I think about it, I never clicked on the end (vertex) of a wire to stretch it with Cadence. But, I did click and drag in the same direction and, if I remember right, it did stretch the wire the way I wanted - that is, the other vertex of the segment stayed put.

Is there any way to get that with Xschem? Here, **when I stretch (CTRL + click & drag), I only want the segment to grow.** Yes, I know that can be accomplished by drawing a wire. With the snap feature, it's not too bad, but, still, the fluid, intuitive click+ drag would be nice to have.

Grazie

[image]

ananthchellappa, 2025-01-26:
The first picture in your previous reply - I never did anything that it Cadence. It doesn't even support it. You always delete a segment and redraw it.  Because the UI is so good, it goes off very smoothly. 

I never had a problem with the UI in Cadence - maybe when you choose a larger font, the forms don't display all of the text, etc. Very minor things. 

I guess we don't need to modify the "click-on-wire-vertex-and-drag-to-start-new-wire" thing, but it would be nice to get a way to do the stretching I would like to have. 

I'm all for minimizing work. Priorities are important. The most important thing for me now is being able to stretch without having to press CTRL and without having to first select an object. That's where Cadence excelled. Granted, they took more than one programmer to build Virtuoso Composer, but having tried that, Tanner S-Edit, LTspice, Qspice, etc, it really has no equal.

StefanSchippers, 2025-01-26:
If you enable Options-> Auto Join / Trim wires, the behavir gets very similar

[image]

closed 2025-05-05 by StefanSchippers