StefanSchippers/xschem #290: Is there a way to bind CTRL + D to deselect-using-click-and-drag rectangle?

ananthchellappa, 2025-01-27:
This is not an "issue" that needs to be fixed, but wondering if this can be done and for some guidance regarding the code.

It arises because I use KDE style bindkeys and therefore ALT + Mouse to drag/resize windows.

Setting plain 'd' to deselect one instance is easy (is there a way to remove the infix from it so that one has to click? Not a big deal, but, again, would like to learn)

I see this

```
 /* Alt - Button1 click to unselect */
 else if(button==Button1 && (SET_MODMASK) ) {
   xctx->last_command = 0;
   xctx->mx_save = mx; xctx->my_save = my;
   xctx->mx_double_save=xctx->mousex_snap;
   xctx->my_double_save=xctx->mousey_snap;
   select_object(xctx->mousex, xctx->mousey, 0, 0, NULL);
   rebuild_selected_array(); /* sets or clears xctx->ui_state SELECTION flag */
 }
```

In callback.c. Because it depends on Button1 already being pressed, what can one do if we want to change to CTRL + D which now waits for a click+drag.

Again, not a real issue needing to be fixed, but, looking to learn about the code base.

Grazie

StefanSchippers, 2025-01-27:
The big issue that needs to be addressed is to add a translation layer in the event system to allow easy user customization of bindkeys.

the **WIndows** key (if present) can be used instead of **Alt** for all command sthat require Alt modifier.

ananthchellappa, 2025-02-06:
Feature available in latest release (SHIFT-'d' for deselection using rectangle. CTRL-'d' now for delete-file. Thank you Stefan.

closed 2025-02-06 by ananthchellappa