StefanSchippers/xschem #292: Implementation of snap cursor and snapping in wire-draw mode

ananthchellappa, 2025-01-29:
Please consider merging this feature from this repo as it provides the same user experience as Cadence ('s' to snap. CTRL-r now for run simulation): 

https://github.com/TheSUPERCD/xschem

The changes are described here : https://docs.google.com/document/d/1ETLK9iiXGWEOvRR4gzMR80oQV1wyVk8h4_TlROCo8t0/edit?tab=t.0

[image]

StefanSchippers, 2025-01-30:
I have created the `SuperChayan` branch in addition to `master` in my repo. 
This is a merge of @TheSUPERCD repository.

I have added a couple of small fixes Chayan could look at (double to int conversion  in `redraw_w_a_l_r_p_z_rubbers()` ) and a couple of variable declaration intermixed with code in `callback.c:1505`, just moved at beginning. 
As long as possible I would keep the source code ANSI C (C89- C90) compatible. 

for merging into my master tree I suggest 2 things:

- Default value of existing or new options (like `infix_interface` and `orthogonal_wiring`) must not change existing behavior, because this will impact existing users. I don't think this is a problem since you can easily keep this setting as you like in xschemrc or even in the system xschemrc (/usr/local/share/xschem/xschemrc). So the default for `infix_interface` should be 1 and for the new `orthogonal_wiring` should be 0.
- For the same reason keybinding defaults should not be changed. You can easily create a `cadence_compat` tcl variable to switch-on  's' and 'Ctrl-r' variations (and any other you like). @TheSUPERCD I suggest this approach to allow a seamless merge, without existing users throwing rotten eggs or stones at me.

For wiring in persistent mode, if a wire placement is in progress the first escape clears the last wire placement, but wire insert mode is still active. A second escape clears the command completely. This is consistent with other commands, for example if you select some objects and press 'm'/'c' to move/copy and you press Escape before confirming the move/copy operation the move is cancelled, but the selection is still active. A second Escape clears the selection.
This is very useful when copying or moving a large portion of a circuit. You select a lot of things, but while moving/copying you realize some elements were not added to the selection. You press escape, add the missing elements to the selection and start with the move/copy again. The same if you selected too many elements, press Esc, unselect some objects and start over again with move/copy.

I am keeping the SuperChayan branch updated and when the changes will not pull the carpet under existing user feets I can apply the merge.

Thank you.
@TheSUPERCD, understanding the xschem code is really not easy, so congrats for the massive work.

ananthchellappa, 2025-01-30:
Sincerely appreciate this Stefan. We'll implement your suggestions.
The variables to keep default behavior the same was an easy one we should have taken care of already. 

Re congrats to Chayan, I tried myself, but couldn't get anywhere, so I was lucky to find him 😊 I know what you mean.

We should keep up the momentum. If it goes well, we can look into raising funds to get open-access. But, before that, we should give some thought to refactoring the code so more developers can get involved easily. "SOLID" principles are what I am thinking of - SRP - single responsibility, OCP - Open-for-extension/Closed for Modification, LIskov substitution (may not apply), Interface Segregation - may not apply, Dependency Inversion - could be considered..

closed 2025-03-09 by StefanSchippers