StefanSchippers/xschem #294: Xschem chrash while opening ihp-sg13g2/libs.tech/xschem/sg13g2_tests_xyce/mc_res_op.sch'

olisnr, 2025-01-30:
on my terminal it says:

`
ich@P43s:~$ xschem '/home/ich/share/pdk/ihp-sg13g2/libs.tech/xschem/sg13g2_tests_xyce/mc_res_op.sch' 
Using run time directory XSCHEM_SHAREDIR = /usr/local/share/xschem
Sourcing /usr/local/share/xschem/xschemrc init file
Sourcing /home/ich/.xschem/xschemrc init file
SG13G2_MODELS: /home/ich/share/pdk/ihp-sg13g2/libs.tech/ngspice/models
SG13G2_MODELS_XYCE: /home/ich/share/pdk/ihp-sg13g2/libs.tech/xyce/models
rename dir (null) to /tmp/xschem_emergencysave_mc_res_op_ccggfcecce failed
EMERGENCY SAVE DIR: /tmp/xschem_emergencysave_mc_res_op_ccggfcecce

FATAL: signal 11
while editing: mc_res_op
`

StefanSchippers, 2025-01-30:
Yes, confirmed. Working on this...

StefanSchippers, 2025-01-30:
Can you please test? fixed in:
https://github.com/StefanSchippers/xschem/commit/60d180de33181e8463072dc1ece9e39a4a0b62c7

Thank you for pointing out this serious bug.

olisnr, 2025-01-30:
super! thanks a lot!!!

closed 2025-01-30 by olisnr