StefanSchippers/xschem #297: Refactoring of event handling code for readability

ananthchellappa, 2025-02-01:
The problem is, when looking at a portion of the code, we don't know which event type we are handling. An example is ButtonPress - the case statement block is so long.

Is it possible to refactor along these lines using separate appropriately-named functions? Since there is only one call to each new function, I am sure the compiler will make the final machine code look the same as the original code.

```
switch (event) {
  case KeyPress:
    handle_key_press(event);
    break;
  case KeyRelease:
    handle_key_release(event);
    break;
  case ButtonPress:
    handle_button_press(event);
    break;
  case ButtonRelease:
    handle_button_release(event);
    break;
  case Expose:
    handle_expose(event);
    break;
  deult:
    dbg(1, "callback(): Event:%d\n",event);
    break;}
```

StefanSchippers, 2025-02-02:
yes this is one of the goals.
To avoid passing too many parameters to the handle functions I guess some callback variables must be moved into the context structure **xctx**, if they are not already there.

Also this refactoring must be done very carefully. The callback.c is a minefield. it's extremely easy to break up everything with apparently innocuous changes. Changes must be done incrementally with care and lot of regression testing.

Since xschem handles multiple schematic tabs or windows global and static variables are essentially forbidden except in special cases. All stateful data is moved into a **xctx** struct. When switching to another tab the **xctx** context is switched too, so all tabs are isolated and do not interfere. This makes it possible to work even on different PDKs within the same xschem process.

ananthchellappa, 2025-02-02:
Was going to add that such refactoring, in this case, only helps if one is using an IDE that shows you which function you are in 😊 VS Code does..

StefanSchippers, 2025-02-02:
Avoiding such giant functions is a good thing to do anyway. The callback routine has grown too big over time.

StefanSchippers, 2025-02-10:
@ananthchellappa, Your changes are merged in.

I have 3 branches in my github xschem repo:
```
- master (git@github.com:StefanSchippers/xschem.git)
- ananthchellappa (fetch of https://github.com/ananthchellappa/xschem.git)
- SuperChayan (fetch of 	https://github.com/TheSUPERCD/xschem.git)
```
Do you think @TheSuperCD can merge your callback.c changes into his fork? some manual intervention is necessary, automatic merge has conflicts.
When he is done I can sync up my copy of TheSuperCD.

ananthchellappa, 2025-02-11:
Yes, we'll look into it. He seems to be busy with his regular job at the moment.

ananthchellappa, 2025-02-11:
Hi Stefan,
        I created a merge request after forking your latest (from 2 hours ago) and copying changes from his latest, using variables to ensure default behavior is not changed.

Added a cadence_style_rc file that is:

# run with $> xschem --script /path/to/cadence_style_rc
set draw_crosshair 1
set crosshair_size 2
set infix_interface 0
set cadence_compat 1
set orthogonal_wiring 1
set snap_cursor 1
set snap_cursor_size 4
set persistent_command 1
set use_cursor_for_selection 1

Tests fine. Hope you are okay with merging.

Thanks,
Ananth

- master ***@***.***:StefanSchippers/xschem.git)

StefanSchippers, 2025-02-12:
merged into master. Thank you.

StefanSchippers, 2025-02-12:
the cadence_style_rc is now part of the install tree and make install will put it in .../share/xschem/ like all other runtime / configuration files.

StefanSchippers, 2025-02-12:
If I remember well Cadence also had by default what in xschem is called Auto Join/Trim wires (xschemrc autotrim_wires) This will join and trim wires: abutting and parallel wires are joined into a single line. 
Adding a non parallel wire that touche an existing wire in an internal point will break the wire in the internal point. You may try this (Options->Auto Join/trim wires)

ananthchellappa, 2025-02-12:
Hi Stefan,
         Latest repo still gives me these :

analog@DESKTOP-O264A5N xschem $ git pull origin master
remote: Enumerating objects: 9, done.
remote: Counting objects: 100% (9/9), done.
remote: Compressing objects: 100% (1/1), done.
remote: Total 5 (delta 4), reused 5 (delta 4), pack-reused 0 (from 0)
Unpacking objects: 100% (5/5), 648 bytes | 36.00 KiB/s, done.
From https://github.com/StefanSchippers/xschem
 * branch              master     -> FETCH_HEAD
   bc293a1d..95cbd807  master     -> origin/master
Updating bc293a1d..95cbd807
Fast-forward
 src/callback.c | 6 ++++++
 src/xschem.tcl | 4 +++-
 2 files changed, 9 insertions(+), 1 deletion(-)

gcc -c -pipe -O2 -I/usr/include/cairo -I/usr/include/libpng16 -I/usr/include/freetype2 -I/usr/include/pixman-1 -I/usr/include/tcl8.6  -o eval_expr.o eval_expr.c
eval_expr.c: In function ‘yyparse’:
eval_expr.c:1109:16: warning: implicit declaration of function ‘yylex’ [-Wimplicit-function-declaration]
 1109 |       yychar = yylex ();
      |                ^~~~~
eval_expr.c:1322:7: warning: implicit declaration of function ‘yyerror’; did you mean ‘yyerrok’? [-Wimplicit-function-declaration]
 1322 |       yyerror (YY_("syntax error"));
      |       ^~~~~~~
      |       yyerrok
eval_expr.y: In function ‘kklex’:
eval_expr.y:218:7: error: ‘kklval’ undeclared (first use in this function); did you mean ‘yylval’?
  218 |       kklval.c = c;
      |       ^~~~~~
      |       yylval
eval_expr.y:218:7: note: each undeclared identifier is reported only once for each function it appears in
eval_expr.y: In function ‘eval_expr’:
eval_expr.y:275:3: warning: implicit declaration of function ‘kkparse’; did you mean ‘yyparse’? [-Wimplicit-function-declaration]
  275 |   kkparse();
      |   ^~~~~~~
      |   yyparse
make[1]: *** [Makefile:135: eval_expr.o] Error 1
make[1]: Leaving directory '/home/analog/Downloads/xschem/src'
make: *** [Makefile:2: all] Error 2

Can we put the suggested compiler settings into the Make setup? Or maybe we can have a README for contributors?

ananthchellappa, 2025-02-12:
I agree Stefan, we can add

Set autotrim_wires 1

To the cadence_style_rc

Thanks!

Adding a non parallel wire that touches an existing wire in an internal point will break the wire in the internal point. You may try this (Options->Auto Join/trim wires)

ananthchellappa, 2025-02-12:
OK, not getting the errors anymore.. I'm able to get a clean build..

BTW, Chayan believes this portion (in handle_double_click) should NOT be inside an if

     if( cadence_compat ) {
      redraw_w_a_l_r_p_z_rubbers(1);
      start_wire(mx, my);
     }

StefanSchippers, 2025-02-12:
Run ./configure again

ananthchellappa, 2025-02-12:
Yes, that helped. I was only doing "make clean".

StefanSchippers, 2025-02-12:
I will check the code. I think Super Chayan is right...
 Should probably be just in the orthogonal wiring id() ....

closed 2025-02-24 by StefanSchippers