StefanSchippers/xschem #302: plots synch X-axis over different plot-types

olisnr, 2025-02-07:
as example, i have a ac and a trans simulation. if i `cntr-f` on of them, the other is also changed. i have to `Unlock X-axis` in both plots to see both simulations.

i think before the X-axis lock was simulation depended. may be im wrong in that, but it would be better anyway.

georgtree, 2025-02-08:
I would add that it would be great, as well as possibility to select which plot are synchronized and which are not.

StefanSchippers, 2025-02-10:
when you zoom /move the xaxis range of a graph all other graphs that are "Locked" will be moved too.
Unlocked graphs will not be touched.

If you set all graphs to "unlocked" you can move/zoom a selected set:
select the graphs you want to move together (click close to the inner border of the graph to select, more can be added by using the Shift key + click). If you zoom / move a graph the selected ones will do too.

olisnr, 2025-02-10:
thats great, but do You think its a good, to lock a frequency x-axis to a time or temp axis? wouldnt it be better to synch only the same type of data? or at least simulation?

StefanSchippers, 2025-02-10:
Sure. I need to add that....

StefanSchippers, 2025-02-10:
It is now done...

https://github.com/user-attachments/assets/c20cee70-c97d-48bf-994e-e0b19c9487c3

olisnr, 2025-02-10:
great! thanks a lot!

closed 2025-02-10 by olisnr