StefanSchippers/xschem #303: Elements preview window scaling

georgtree, 2025-02-08:
Hello!
Is it possible of calculate boundaries or element in preview window without taking into account the hidden text?
Now it is scales with full boundaries:

[image]

And if turn on Show hidden text:

[image]

Thank you!

StefanSchippers, 2025-02-09:
Last commit addresses this issue. If symbol contains hidden texts and xschem is configured to not display hidden texts a full zoom will not include this texts in the bounting box.

Please check.

georgtree, 2025-02-09:
Just tested - if show hidden text enabled, it shows it only after second click to the same element in the list.

StefanSchippers, 2025-02-09:
Ouch... you are right. It's incredible how things can go wrong...

StefanSchippers, 2025-02-09:
ok, this time it seems working.

georgtree, 2025-02-09:
Yes, now it works, just tested, thank you!

closed 2025-02-09 by georgtree