StefanSchippers/xschem #304: Debug messages with custom elements

georgtree, 2025-02-08:
hello!
When I generate symbol with tcl script, I got such messages in console:
```
% SKIP RECORD
exec tclsh "$0" "$@"
END SKIP RECORD
SKIP RECORD
set in_string [lindex $argv 0]
END SKIP RECORD
SKIP RECORD
if {$in_string eq {} } {set in_string "650 0.967 0.502560 0.039964 0.188070 0.000781 42.2283 50 2.2e6"}
END SKIP RECORD
SKIP RECORD
set params [split $in_string " "]
END SKIP RECORD
```

As I understand, this is a message from symbol reader that skips unrecognized lines. But how to turn off them off? 
Thank you.

StefanSchippers, 2025-02-08:
Yo did something wrong. You are trying to read a tcl script as a xschem file (symbol or schematic)
xschem does not recognize any of these lines (it is tcl code not an xschem file)  and skips everything.

georgtree, 2025-02-08:
Why? It is tcl a script that generates the symbol with substitution of some variables.

StefanSchippers, 2025-02-08:
What is the sequence of operations you are doing?
from the shell? 
inside the xschem console?

StefanSchippers, 2025-02-08:
the SKIP RECORD is a message printed when a unrecognized line is read as an xschem file:

test.sch:
```
aaa
bbb
```
$ xschem test.sch
```
SKIP RECORD
aaa
END SKIP RECORD
SKIP RECORD
bbb
END SKIP RECORD
```

georgtree, 2025-02-08:
I call
```tcl
xschem place_symbol
```
from my script

StefanSchippers, 2025-02-08:
`xschem place_symbol` runs `xschem` with `place_symbol` as a schematic to open.

you can run the script with tclsh:

`tclsh place_symbol`

if the script contains _xschem specific commands_ then use:

`xschem --script place_symbol`

georgtree, 2025-02-08:
I call this command from bind script:
```tcl
$globVars(treeviewPath) tag bind element_gen_nin <Double-1> {
    global globVars
    set inputVar 2
    set top [toplevel .top]
    label $top.lb -text [dict get $dictEntryMessage [$globVars(treeviewPath) focus]]
    entry $top.eb -textvariable inputVar
    button $top.bu -command [list incr ${top}done] -text OK
    pack {*}[winfo children $top]
    vwait ::${top}done
    unset -nocomplain ::${top}done
    destroy $top
    xschem place_symbol [$globVars(treeviewPath) focus]($inputVar)
}
```
What to do in that case?

georgtree, 2025-02-08:
I run this command from the bind script:
```tcl
$globVars(treeviewPath) tag bind element_gen_nin <Double-1> {
    global globVars
    set inputVar 2
    set top [toplevel .top]
    label $top.lb -text [dict get $dictEntryMessage [$globVars(treeviewPath) focus]]
    entry $top.eb -textvariable inputVar
    button $top.bu -command [list incr ${top}done] -text OK
    pack {*}[winfo children $top]
    vwait ::${top}done
    unset -nocomplain ::${top}done
    destroy $top
    xschem -script place_symbol [$globVars(treeviewPath) focus]($inputVar)
}
```

What to do in that case?
 xschem --script return error - no such command

StefanSchippers, 2025-02-08:
Sorry I don't know what you are trying to do.
Please give the context.

- Are you writing your own application in tcl and binding some actions to a mouse event?
- Are you working inside xschem and adding some mouse bindings?

StefanSchippers, 2025-02-08:
because outside xschem `xschem` means launch the xschem executable
inside xschem `xschem` is an extended tcl command.

georgtree, 2025-02-08:
I am writing the plugin with library tree and load it in xschem interpreter after xschem startup, then I place the components with double click:

[image]

StefanSchippers, 2025-02-08:
Ok, Now I understand better.

I tried the commands below to place components. `xschem place_symbol` requires user intervention to decide where to place the component.

the `xschem instance` command accepts also the position so no user intervention is required.

https://github.com/user-attachments/assets/7d9bd8cd-9236-4b1b-a811-6a0f7732ff7e

georgtree, 2025-02-08:
Yes right, so when I place symbol which is effectively a pcell, it throws errors to the terminal

StefanSchippers, 2025-02-08:
Try to look into the .sym file you are placing and see if it is a clean xschem file.

StefanSchippers, 2025-02-08:
This message:
% SKIP RECORD
exec tclsh "$0" "$@"
END SKIP RECORD

means the .sym file you are placing contains a line  like:
exec tclsh "$0" "$@"
that is clearly not part of an xschem file.
xschem does not recognize such a line and skips it altogether.

georgtree, 2025-02-08:
Yes right, because it is not a sym file but tcl script which generate the sym file at the call:
```tcl
#!/bin/sh
# the next line restarts using wish \
exec tclsh "$0" "$@"

set in_string [lindex $argv 0]
if {$in_string eq {} } {set in_string "650 0.967 0.502560 0.039964 0.188070 0.000781 42.2283 50 2.2e6"}
set params [split $in_string " "]

puts "v {xschem version=3.4.5 file_version=1.2}
G {}
K {type=magnetic_flux_tubes
format=\"@name @@n1 @@n2 @subckt HSAT=@HSAT M=@M R=@R Q=@Q P1=@P1 P2=@P2 HC=@HC K=@K EDDY=@EDDY SIGMA=@SIGMA D=@D L=@L A=@A MAGREL=@MAGREL\"
template=\"name=X1 subckt=tellinen_everett_hyst_core EDDY=0 L=0.01 A=0.01 MAGREL=0.0 D=0.5e-3 HSAT=[lindex $params 0] M=[lindex $params 1] R=[lindex $params 2] Q=[lindex $params 3] P1=[lindex $params 4] P2=[lindex $params 5] HC=[lindex $params 6] K=[lindex $params 7] SIGMA=[lindex $params 8]\"}
V {}
S {}
E {}
L 16 -30 0 -20 0 {}
L 16 20 0 30 0 {}
L 16 2.5 -5 5 -5 {}
L 16 -5 5 -2.5 5 {}
B 5 -32.5 -2.5 -27.5 2.5 {name=n1 dir=inout class=flux_tubes}
B 5 27.5 -2.5 32.5 2.5 {name=n2 dir=inout class=flux_tubes}
B 16 -20 -10 20 10 {fill=0}
P 16 4 -5 5 -2.5 5 0 -5 2.5 -5 {bezier=1}
P 16 4 -2.5 5 0 5 2.5 -5 5 -5 {bezier=1}
T {@subckt} -20 -30 0 0 0.15 0.15 {hide=true}
T {@name} -20 -20 0 0 0.15 0.15 {}
T {L=@L 
A=@A
EDDY=@EDDY
MAGREL=@MAGREL} -20 10 0 0 0.1 0.1 {}
T {HSAT=@HSAT 
M=@M 
R=@R 
Q=@Q 
P1=@P1 
P2=@P2
HC=@HC 
K=@K 
SIGMA=@SIGMA
D=@D} -20 37.5 0 0 0.1 0.1 {hide=true}
T {TE} 7.5 0 0 0 0.1 0.1 { layer=16}
"

```

StefanSchippers, 2025-02-08:
if the file is a generator script call it with this syntax:

xschem place_symbol generator.tcl()

add the parenthesis, so xschem understand it is a generator and it executes the script (instead of loading) and loads the standard outpt produced by the scripts. If the script takes arguments call it as:

xschem place_symbol {generator.tcl( arg1 , arg2)}

georgtree, 2025-02-08:
It calls like that:
```tcl
xschem place_symbol [$globVars(treeviewPath) focus](\"[dict get $data $selectedVal]\")
```

StefanSchippers, 2025-02-08:
I tried these examples:

https://github.com/user-attachments/assets/474cbcbd-8fd2-489d-b786-566e8937d92d

georgtree, 2025-02-08:
Hm, strange, I go the same route and still get the warnings:

[image]

georgtree, 2025-02-08:
i tried to do it from console, and it does not issue warnings...

StefanSchippers, 2025-02-08:
There is an internal procedure in xschem, `is_xschem_file`. This procedure returns GENERATOR if the file is recognized as a generator:

```
% is_xschem_file /home/schippes/gen.tcl
GENERATOR
%
```

this function is called automatically by `xschem place_symbol` to decide what to do.
for some reason the heuristic fails and xschem loads the file instead of executing it.

for scripts the pattern that is checked to decide what to do is the initial shebang `!#` on first line.

georgtree, 2025-02-08:
Maybe there could be added the explicit option like `-script` to set the type?

StefanSchippers, 2025-02-08:
yes this can be added, but it is more important to understand why it is not recognized as a generator, since the first line correctly start with #!. There is something else that is not working here.

StefanSchippers, 2025-02-08:
George, I think I found the problem.

the xschem place_symbol expects a file, when calling it with a generator and arguments
(like `xschem place_symbol /path/to/gen.tcl(par1, par2)` it tries to find the given filename (including the parenthesis and args) and of course it does not find it, so it can not deduce if it is a script or file to load. WIll fix that.

StefanSchippers, 2025-02-09:
Can you please test with latest commit? I have fixed the above mentioned problem.

georgtree, 2025-02-09:
I checked, messages are the same. But generator does it's job, so it looks like xschem understands that it is generator and executes it, but still issues the messages

StefanSchippers, 2025-02-09:
Can you extract the complete string that is given to xschem place_symbol ?
(generator and parameters)

georgtree, 2025-02-09:
The same generator from the comment above produces:
```
% SKIP RECORD
exec tclsh "$0" "$@"
END SKIP RECORD
SKIP RECORD
set in_string [lindex $argv 0]
END SKIP RECORD
SKIP RECORD
if {$in_string eq {} } {set in_string "650 0.967 0.502560 0.039964 0.188070 0.000781 42.2283 50 2.2e6"}
END SKIP RECORD
SKIP RECORD
set params [split $in_string " "]
END SKIP RECORD
SKIP RECORD
puts "v {xschem version=3.4.5 file_version=1.2}
END SKIP RECORD
SKIP RECORD
"END SKIP RECORD
```

The input is: `.sym_links/phys.mag.flux_tubes.shapes.hyst_magn/tellinen_everett_hyst_core_th.tcl("1000 0.919590 0.728070 0.015757 0.016407 0.001049 98.0842 110 2.56e6")`

The code that calls it:

`xschem place_symbol [$globVars(treeviewPath) focus](\"[dict get $data $selectedVal]\")`

StefanSchippers, 2025-02-09:
try to wrap the entire filename+args  into " and ":
`xschem place_symbol "[$globVars(treeviewPath) focus](\"[dict get $data $selectedVal]\")"`

georgtree, 2025-02-09:
Nope, still the same issue

StefanSchippers, 2025-02-09:
I have created the dictionaries, array vars, and procedures to emulate the command and in the console the following:

`xschem place_symbol [$globVars(treeviewPath) focus](\"[dict get $data $selectedVal]\")`

works just fine with no SKIP RECORD messages...

StefanSchippers, 2025-02-09:
I have also bound the command to a button:

`button .top.ins  -command "xschem place_symbol \{[$globVars(treeviewPath) focus](\"[dict get $data $selectedVal]\")\}"
`
Note the various escaped brackets and quotes to get around the infamous tcl _quoting hell_.

Clicking the button places the symbol with al the parameters assigned.

Tcl purists avoid using escaped quotes and brackets and use the [list ....] commands, but I never got anything working with this approach. It works only in the books.

georgtree, 2025-02-09:
I tried this:
```
eval [list xschem place_symbol [$globVars(treeviewPath) focus](\"[dict get $data $selectedVal]\")]
```
But issue is still persist

StefanSchippers, 2025-02-09:
to simplify the problem create a toplevel window with a button:
```
toplevel .top
button.top.ins -text Insert -command .....
pack .top.ins
```

and try all combination of escapes in the -command until you get the symbol added with no error messages.

georgtree, 2025-02-09:
Interesting - button doesn't issue the warnings:

```tcl
toplevel .top
wm title .top Selection
button .top.ins -text Insert -command {xschem place_symbol ".sym_links/phys.mag.flux_tubes.shapes.hyst_magn/tellinen_everett_hyst_core_th.tcl(\"1000 0.919590 0.728070 0.015757 0.016407 0.001049 98.0842 110 2.56e6\")"}
pack .top.ins
```

georgtree, 2025-02-09:
The only difference I see is the usage of escaping for " symbol

georgtree, 2025-02-09:
I found the issue - it goes not from the place_symbol, but from the preview window.

georgtree, 2025-02-09:
This fixes the issue;
```tcl
    if {[file extension [$globVars(treeviewPath) focus]] eq ".tcl"} {
        xschem preview_window draw $globVars(previewPath) [$globVars(treeviewPath) focus]()
    } else {
        xschem preview_window draw $globVars(previewPath) [$globVars(treeviewPath) focus]
    }
```

closed 2025-02-09 by georgtree

StefanSchippers, 2025-02-09:
Ah ok, this is interesting. I didn't pay much attention for displaying generators in preview window. I will add something similar in the load file previewer. 

Generators should always produce a valid symbol if called without arguments (like gen.tcl()) , by assuming reasonable defaults for parameters.

A more general approach is :
```
    if {[is_xschem_file [$globVars(treeviewPath) focus]] eq {GENERATOR} } {
        xschem preview_window draw $globVars(previewPath) [$globVars(treeviewPath) focus]()
    } else {
        xschem preview_window draw $globVars(previewPath) [$globVars(treeviewPath) focus]
    }
```

StefanSchippers, 2025-02-09:
Ok now xschem accepts a filename in the form of `/path/to/gen.tcl` or `/path/to/gen.tcl()` and loads the generator.
Parenthesis are added if the file is recognized as a generator. The above fix should not be needed anymore,