StefanSchippers/xschem #305: Setup custom colors for highlightning

georgtree, 2025-02-08:
Hello!
How could i set colors for highlighting, like you showed here:
https://github.com/StefanSchippers/xschem/issues/197#issuecomment-2122191408

I tried the hex values, but it doesn't work. I see there is some digit code, and assume it connected to layers number, but it looks like the correspondence is not direct (layer number -> color number).

Thank you!

georgtree, 2025-02-08:
I found that shift is 7, so highlighting works as expected, BUT I can't highlight with the color of 4 level...I see that it is expected, but what if I still want to use the color of 4 layer?

StefanSchippers, 2025-02-09:
the `xschem set_hilight_color` maps color 0... to layer 7...
This was done to avoid using layers 0...6 for hilights since these are the most used for drawing schematics.

You can override this mapping by using negative numbers:

```
xschem set_hilight_color -4  ;# uses layer 4
xschem set_hilight_color -1  ;# uses layer 1

xschem set_hilight_color 0 ;# uses layer 7
xschem set_hilight_color 1 ;# uses layer 8
```

closed 2025-02-09 by georgtree

georgtree, 2025-02-10:
Thank you again, now highlighting works great (xschem the only such a tool that allows such a flexibility, thank you for it):

[image]