StefanSchippers/xschem #306: sometimes i have cursor redrawing problems after PDF export

olisnr, 2025-02-09:
its not so easy to reproduce: if i exprort PDFs using `File :: Image Exprort :: Hirarchical PDF/EPS export` the main window has problems with updateing and looks like this, after moving the cursor:

[image]

if i change the window-size its working perfekt again.

StefanSchippers, 2025-02-09:
can you please attach the xschemrc file you are using ? so I try with your same settings...

StefanSchippers, 2025-02-09:
I see some garbage like this but for a very short time. Will so some tests...

https://github.com/user-attachments/assets/fc073233-9825-4dd6-badc-a56434c6e6ee

StefanSchippers, 2025-02-09:
If you press 'f' to do a full zoom (and no resizing window)  do you still see garbage when moving the crosshair cursor?

olisnr, 2025-02-09:
https://github.com/user-attachments/assets/88651326-bde9-46c2-add3-2a7b0d3d3fbf

in the end i press the `f`, before solving the problem with changing the window size.

[xschemrc.zip](https://github.com/user-attachments/files/18725929/xschemrc.zip)

StefanSchippers, 2025-02-09:
This morning I did one fix that removed the garbage I saw for a couple of seconds  when doing a hierarchical pdf export. I don't see the garbage anymore, but yor case is different as the garbage remains on screen until you resize or redraw.

Anyway if you have a chance to test again...

olisnr, 2025-02-09:
I'm sorry, but garbage collecting doesn't work in my system.

StefanSchippers, 2025-02-09:
One possibility to investigate is to run a vnc server on the host machine and run xschem inside the vnc session. I often do this when debugging graphical problems. VNC is a very clean implementation of the X11 protocol, this is very useful to separate application bugs from desktop bugs.

https://github.com/user-attachments/assets/4b118310-6948-466f-b9e1-163e740730be

StefanSchippers, 2025-02-09:
Can you also please test doing a hierarchical PDF export on the same schematic but without any graph ?

StefanSchippers, 2025-02-10:
Another interesting test is to try making all graph rectangles smaller. 
graphs are rendered by creating a big bitmap and embedding the encoded bitmap in the postscript file.
Some graphic systems have overflow problems if bitmaps grow too big. 
I have inserted some tests to scale down graphs if they are too big.

olisnr, 2025-02-10:
the same schematic has problems with empty plots, no data loaded.
the same schematic has NO problems without plots.

StefanSchippers, 2025-02-10:
Thank you very much for the updates :-)

olisnr, 2025-02-10:
the new version has the same problem even without data

olisnr, 2025-02-10:
should i try some other windowmanager or videodriver?

StefanSchippers, 2025-02-10:
I don't think the window manager is the culprit. 
What GPU are you using ? (sorry you told me but forgot)

olisnr, 2025-02-10:
Graphics:
  Device-1: Intel WhiskeyLake-U GT2 [UHD Graphics 620] vendor: Lenovo driver: i915 v: kernel
    ports: active: HDMI-A-2,eDP-1 empty: DP-1,DP-2,HDMI-A-1 bus-ID: 00:02.0 chip-ID: 8086:3ea0
    class-ID: 0300
  Device-2: NVIDIA GP108GLM [Quadro P520] vendor: Lenovo driver: nvidia v: 535.183.01 pcie:
    speed: 2.5 GT/s lanes: 4 bus-ID: 3c:00.0 chip-ID: 10de:1d34 class-ID: 0302

nvidea

StefanSchippers, 2025-02-10:
I have almost same setup: 
```
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 620 (rev 02)
01:00.0 3D controller: NVIDIA Corporation GM108M [GeForce 930MX] (rev a2)
```

I have the following /etc/X11/xorg.conf file:

```
Section "InputClass"
   Identifier   "Touchpad"
   Driver       "libinput"
   Option "Tapping" "true"
   Option "TappingButtonMap" "lmr"
   Option "TappingDragLock" "true"
   Option "TappingDrag" "true"
   Option "AccelSpeed" "0.6"
   Option "MiddleEmulation" "true"
EndSection

# best for video playing
Section "Device"
  Identifier "intel1"
  Driver "intel"
  # Option "AccelMethod" "sna"
  Option "VSync" "true"
  Option "TearFree" "true"
  Option "TripleBuffer" "true"
  Option "SwapbuffersWait" "false"
  Option "DRI" "iris"
EndSection

# good for video playing
#Section "Device"
#   Identifier  "intel2"
#   Driver      "modesetting"
#   Option "DRI" "iris"
#EndSection

# best for 2D graphics (Xlib)
Section "Device"
   Identifier  "intel3"
   Driver      "intel"
   Option "PageFlip" "false"
   Option "DRI" "iris"
EndSection

Section "Screen"
   Identifier   "screen_intel"
   Device       "intel1"
EndSection
```

I think you can temporarily put this file in  /etc/X11/xorg.conf (save existing one if any!) and try to restart the computer. (I don't think logging out of the desktop is enough, not sure).

You see I have 3 settings for the Intel GPU (intel1, intel2, intel3) and I am currently using intel1 while for 2D operations intel3 is the fastest (specify which one to use in the "Screen" section).

olisnr, 2025-02-10:
i tried a bit more: if the graph is really small, i dont get the error. even if i make a lot of copies

olisnr, 2025-02-10:
now i try Your config file. expect my next message i 1..2 years, when i got the system running again :)

olisnr, 2025-02-10:
`/etc/X11/xorg.conf` didnt change anything

StefanSchippers, 2025-02-10:
Thank you for your patience. Looks a difficult problem.

olisnr, 2025-02-10:
there is an other observation: if the window is about y > 2 * x there is no problem. or the problem would be outside the window that is drawn

StefanSchippers, 2025-02-10:
One thing you can test is this in the xschem console:

`% xschem resetwin 1 1 0 0 0 ; xschem redraw`

Does this fix the garbage without doing a window resize?

olisnr, 2025-02-10:
yes

StefanSchippers, 2025-02-10:
I can add the two above statements after a hierarchy pdf export. Not a clean solution, but a workaround.

olisnr, 2025-02-10:
i wouldnt. its not really a problem for working with Xschem. and if You do it, a problem could be hidden until it breaks something other.

StefanSchippers, 2025-02-10:
I do some cleanup of the postscript export code. May be I find something...

StefanSchippers, 2025-02-10:
I was able to see the problem. I had to make the schematic window very big vertically, using two monitors. The cursor garbage in on the lower part of the image , which is on the second monitor (the laptop screen, below the primary monitor).

[image]

StefanSchippers, 2025-02-10:
Now I can work for a fix :-)

closed 2025-02-10 by olisnr